adafruit/circuitpython #1437: super object has no attribute __new__

Fuermohao, 2019-01-02:
```python
>>> class A:
...     def __new__(cls):
...         super(A, cls).__new__(cls)
...         
...         
... 
>>> a = A()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 3, in __new__
AttributeError: 'super' object has no attribute '__new__'
```
Is this feature implemented?

notro, 2019-01-02:
```__new__``` requires [MICROPY_CPYTHON_COMPAT](http://elixir.tronnes.org/circuitpython/latest/ident/MICROPY_CPYTHON_COMPAT) which is only enabled on samd51/M4 builds (maybe  on NRF but definitely not samd21/M0) due to the increased flash usage. See #1167.

ladyada, 2019-01-02:
@dhalbert @tannewt if either of ya can verify this (i believe its true) then @kattni please add to the Expectations page :)

tannewt, 2019-01-03:
Looks like @notro is correct: http://elixir.tronnes.org/circuitpython/latest/source/py/objobject.c#L71

I'd love to see CPYTHON_COMPAT enabled for all builds. Until then we can add it to expectations though.

dhalbert, 2019-01-03:
In a recent version of `master` right now, a Trinket M0 build with `MICROPY_CPYTHON_COMPAT` off has 5668 bytes free. Turned on, there are 4684 bytes free. So I think we could turn this on everywhere.

tannewt, 2019-01-03:
I think we should sit on that space a bit longer until displayio is further along.

I believe the Arduino Zero in French is the largest build at the moment too. Trinket is smaller because it has fewer pins.

Singein, 2019-01-10:
@ladyada 
> I'd love to see CPYTHON_COMPAT enabled for all builds. Until then we can add it to expectations though.

I do hope this option can be enabled for all builds. Waiting for your new release~

@dhalbert 
> In a recent version of master right now, a Trinket M0 build with MICROPY_CPYTHON_COMPAT off has 5668 bytes free. Turned on, there are 4684 bytes free. So I think we could turn this on everywhere.
I hope that we can release two versions of the firmware, let users choose which one to use.

tannewt, 2019-03-20:
Do we have room to enable COMPAT on more builds for 4.0.0?

dhalbert, 2019-03-20:
Yes, now we do. I'll try it and do a PR.

tannewt, 2019-03-20:
Fixed by #1675. Thanks @dhalbert !

closed 2019-03-20 by tannewt