adafruit/circuitpython #1438: Long Term: port to risc v

TG-Techie, 2019-01-02:
this board is the first embedded risc-v processor but porting would allow for flexibility in the future:
https://www.sifive.com/boards/hifive1

tannewt, 2019-01-03:
The HiFive1 doesn't have native USB so I'd rather not port to it.

However, the FPGA folks will have native USB and RISCV cores. It brings up an interesting point though where we really define a port by peripheral family instead of the cpu core itself. Porting to the RISCV cpu itself shouldn't be hard since its a gcc change. The peripheral support is the hard part.

closed 2019-01-03 by tannewt

TG-Techie, 2019-01-09:
thank you @tannewt  very helpful information about how porting is done as of now, and brain candy.