adafruit/circuitpython #1441: Sort releases into folders by language

TG-Techie, 2019-01-04:
I find it quite hard to locate the correct uf2 for my board because of the length of the list expanding A LOT. It is awesome that more languages have been added! However, I would imagine that if a single person is downloading a uf2 they will probably be looking for their specific language.

kattni, 2019-01-04:
We are working on a downloads page that will simplify the process significantly.

TG-Techie, 2019-01-06:
oh awesome either way!  thank you

tannewt, 2019-01-07:
Sorting them in S3 by language would be nice. Should be a change here: https://github.com/adafruit/circuitpython/blob/master/tools/build_release_files.py

vgoodwinv, 2019-01-15:
Added language above board in directory structure.  Did not remove board directories from being replicated under every language.  So, every language directory has all board directories.  Maybe instead board directories should not be created?

tannewt, 2019-01-15:
Can we put language below board so that we reuse the existing folders? I'm not sure whether they'll end up at the top though.

TG-Techie, 2019-01-15:
If you move {language} infront of the adafruit_foo_bar{}{}{} in the final_filename variablei think it will do what you're asking.    I'm curious why you want board first? I would always to to English then the specific board.    Yo hablo un poco español pero solo usar una lengua cuando estoy coding

tannewt, 2019-01-15:
Done! Thanks @vgoodwinv. https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/feather_m4_express/en_US/

They are listed at the bottom. We'll clean out old intermediate builds soon to make the page shorter.

closed 2019-01-15 by tannewt

ladyada, 2019-01-15:
awesome!