adafruit/circuitpython #1443: Audio doesn't play if class defined

qubist, 2019-01-08:
I'm having a very odd issue. Using the audioio library to play sounds, if I define a class before I play my sound using `mixer.play()` the sound file isn't played. Commenting out the class definition makes the sound play flawlessly.

Here's my code:

```
import board
import audioio

def makeMixer():
    audio = audioio.AudioOut(board.A1, right_channel=board.A0)
    mixer = audioio.Mixer(voice_count=1,
                          sample_rate=22050,
                          channel_count=2,
                          bits_per_sample=16,
                          samples_signed=True)
    audio.play(mixer)
    return mixer

mixer = makeMixer()

sample = audioio.WaveFile(open("/1.wav", "rb"))

# uncomment this class definition and listen to the sound not playing
#
# class foo:
#     def __init__():
#         pass
#

mixer.play(sample)
while mixer.playing:
    pass
```

I'm using a NeoTrellis M4. CircuitPython version: `Adafruit CircuitPython 4.0.0-alpha.5-318-g6773fe6 on 2018-12-12; Adafruit Trellis M4 Express with samd51g19` The CircuitPython drive has a 16 bps, stereo, 22050 sample rate .wav file in it named `1.wav`. There's also the entire latest `lib` bundle.

Any thoughts on what's going on here? Help would be appreciated.

Thanks!

dhalbert, 2019-01-08:
This _might_ be related to #1218. Not clear that the global dictionaries are doing long-lived exactly right.

qubist, 2019-01-11:
I'm having more separate audioio weirdness too, working on a different project on my NeoTrellis M4. Errors being thrown that report back to different lines in the code every time, sometimes lines with only `   else:` on them. I take out audioio library and all the audio functionality of the project, though, and it works flawlessly. I'll narrow it down to a neat example case and post a proper issue soon.

tannewt, 2019-01-11:
@qubist If it doesn't have the class define in it please file a separate issue. Thanks!

ladyada, 2019-01-14:
isnt the `audio` var in makeMixer function has no references to it, its gc'd after the call is complete - so they'd have to make that `audio` object global (defined outside function)?

qubist, 2019-01-14:
> isnt the `audio` var in makeMixer function has no references to it, its gc'd after the call is complete - so they'd have to make that `audio` object global (defined outside function)?

Then this code should work with and without the class defined, right?

```
import board
import audioio

audio = audioio.AudioOut(board.A1, right_channel=board.A0)
mixer = audioio.Mixer(voice_count=1,
                      sample_rate=22050,
                      channel_count=2,
                      bits_per_sample=16,
                      samples_signed=True)
audio.play(mixer)

sample = audioio.WaveFile(open("/1.wav", "rb"))

# uncomment this class definition and listen to the sound not playing
# 
# class foo:
#     def __init__():
#         pass

mixer.play(sample)
while mixer.playing:
    pass
```
I tried it and it still behaves the same way

ladyada, 2019-01-14:
oooh ok - i didnt test it! that's even more concise :)

commit 5555a244 (2019-02-02): Never long live the main dictionary.

It's contents change often and may be referenced elsewhere.

Fixes #1443
closed 2019-02-02 by tannewt