adafruit/circuitpython #1451: Format Strings for struct.pack_into() don't match signed/unsigned vars

ATMakersBill, 2019-01-10:
As we were trying to track down a problem w/the gamepad HID interface (or more likely the Microsoft XAC's implementation) @dhalbert and I saw that the format strings for pack_into() were not taking into consideration whether the variable was signed.  For example in gamepad.py you'll find:

```
        struct.pack_into('<HBBBB', self._report, 0,
                         self._buttons_state,
                         self._joy_x, self._joy_y,
                         self._joy_z, self._joy_r_z)
```

The "<HBBBB" string says that it's expecting an unsigned 16-bit value followed by four unsigned bytes.  However, we are sending signed bytes (-127 through 127) as defined in our descriptor.

In this case, circruitpython doesn't care because it would encode them the same way.  However, CPython would error out after checking this and in general this is dangerous, because with types > 8 bit, we very much do care about endianness.

@dhalbert asked that I open an issue about this.  The fix on gamepad.py is pretty simple, just change the string to "<Hbbbb", however, we should probably do a sanity check on this.

tannewt, 2019-01-10:
I think this should remain here. We should error out when a value is out of bounds.

Adding a second issue to HID would make sense to me though.

godlygeek, 2019-05-06:
I'll take a swing at this...

ladyada, 2019-05-06:
yay!

godlygeek, 2019-05-06:
Also broken:

```
>>> x = 256
>>> x.to_bytes(1, 'big')
b'\x00'
```

This should raise an `OverflowError` instead...

godlygeek, 2019-05-06:
So should:

```
>>> array.array('B', [255, 256])
array('B', [255, 0])
```

godlygeek, 2019-05-13:
@tannewt - I believe this can be closed after #1860 and #1879.

tannewt, 2019-05-13:
Thanks @godlygeek !

closed 2019-05-13 by tannewt