adafruit/circuitpython #1452: Spanish translation update

C47D, 2019-01-10:
Hola @sabas1080 y @yeyeto2788,

Solo para preguntarles si están trabajando en [mantener actualizada] la traducción al español, probablemente pueda retomar el tema la siguiente semana, lo pregunto para no duplicar trabajo.

Saludos.
-c

yeyeto2788, 2019-01-10:
Hola @C47D,

Yo honestamente no lo he mirado pero si no hay nadie disponible puedo verlo entre mañana y este fin de semana. 

Igualmente lo mirare y subiré los cambios a mi repositorio (intentando actualizar este issue).

Saludos.

yeyeto2788, 2019-01-10:
**Quick Update:**

He visto que faltan muchas cosas por traducir, en una vista rápida he visto las siguientes lineas que les faltan traducción:

* [1245](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L1245)
* [1257](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L1257)
* [1261](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L1261)
* [2298](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2298)
* [2306](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2306)
* [2318](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2318)
* [2322](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2322)
* [2326](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2326)
* [2330](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2330)
* [2359](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2359)
* [2371](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2371)
* [2383](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2383)
* [2387](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2387)
* [2430](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2430)
* [2438](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2438)
* [2446](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2446)
* [2548](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2548)
* [2552](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2552)
* [2556](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2556)
* [2560](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2560)
* [2580](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2580)
* [2582](https://github.com/yeyeto2788/circuitpython/blob/5c776db9b113d5a0a70f8e8458c11d22772f9e3a/locale/es.po#L2582)
* Hay mas pero esto mas o menos fue lo que vi rápidamente.

Lo mirare en el tiempo que te comente hace unos minutos, cualquier cosa no duden en contactar conmigo.

Saludos.

sabas1080, 2019-01-10:
No he estado trabajando en eso, ando perdido en makecode, pero cualquier review que necesiten pueden etiquetarme

C47D, 2019-01-11:
@yeyeto2788 Podríamos dividirnos los strings que falta traducir y los que falta actualizar, si quieres revisalo con calma y los que no alcances a actualizar me los dejas, ¿que te parece?.

@sabas1080 Claro, te mantendremos al tanto.

Saludos

sabas1080, 2019-01-11:
Por cierto ando armando sus tarjetas para enviarlas, una disculpa por la tardanza

yeyeto2788, 2019-01-11:
@C47D, @sabas1080 ya lo tienen. Al final he tenido un poco de tiempo hoy y no quise dejarlo pasar.

Revisarlo en lo que puedan y cualquier cosa me dicen.

Saludos.

P.D: @sabas1080 no hay problema por la tardanza, ayer casualmente te escribí un correo 😋. Sigo ansioso de que llegue a mi buzón la placa para poder trastear con ella.

C47D, 2019-01-13:
@yeyeto2788 Gracias por el apoyo!
@sabas1080 Gracias por la tarjeta, la esperamos con ansia por estos lados.

Saludos

closed 2019-01-13 by C47D