adafruit/circuitpython #1454: ugame build failure

jerryneedell, 2019-01-11:
Has ugame been removed? I ran build_release_files.py on the current master and it fails:
```
Build ugame10 for en_US took 24.40s and failed
make: Entering directory '/home/jerryneedell/projects/adafruit_github/circuitpython_master/ports/atmel-samd'
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
make: *** No rule to make target '../../frozen/ugame10', needed by 'build-ugame10/frozen_mpy'.  Stop.
make: Leaving directory '/home/jerryneedell/projects/adafruit_github/circuitpython_master/ports/atmel-samd'

Cannot find file ../ports/atmel-samd/build-ugame10/firmware.uf2

```

dhalbert, 2019-01-11:
It's not built by default, but the board definition is still there. I'm not sure it was ever built by default. It's not in https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin

jerryneedell, 2019-01-11:
Ok, I guess I did not notice the errors before. No problem..

closed 2019-01-11 by jerryneedell