adafruit/circuitpython #1456: I2C not working on Grand Central

jerryneedell, 2019-01-11:
I cannot get my Grand Central to communicate with some I2C devices -- tried FRAM and LSM9DS1
I am using SDA = D20, SCL = D21 as per:
https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/boards/grandcentral_m4_express/pins.c#L61
for the FRAM
```

Adafruit CircuitPython 4.0.0-alpha.5-120-g573c6d60b on 2019-01-11; Adafruit Grand Central M4 Express with samd51p20
>>> import fram_i2c_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "fram_i2c_simpletest.py", line 9, in <module>
  File "/lib/adafruit_fram.py", line 209, in __init__
  File "/lib/adafruit_bus_device/i2c_device.py", line 78, in __init__
  File "/lib/adafruit_bus_device/i2c_device.py", line 76, in __init__
ValueError: No I2C device at address: 7c
>>> 
```

if I run I2C_SCAN, it just hangs -- cannot control C out -- have to hard reset.
i2c_scan.py
```
import board
import busio

i2c = busio.I2C(board.SCL, board.SDA)
count = 0

# Wait for I2C lock
while not i2c.try_lock():
    pass

# Scan for devices on the I2C bus
print("Scanning I2C bus")
for x in i2c.scan():
    print(hex(x))
    count += 1

print("%d device(s) found on I2C bus" % count)

# Release the I2C bus
i2c.unlock()

```

ladyada, 2019-01-11:
try the SDA/SCL pins?

jerryneedell, 2019-01-11:
I did  those first -- they did not work either.

jerryneedell, 2019-01-11:
did you test with CP or arduino?

ladyada, 2019-01-11:
wierd - just arduino that i can recall

jerryneedell, 2019-01-11:
I can try arduino just to make sure HW is OK , but I think there is a problem in CP -- there was a forum post as well https://forums.adafruit.com/viewtopic.php?f=60&t=146073

ladyada, 2019-01-11:
hmm - ok wanna take a look at the board definition? see if maybe the pins aren't defined right?

jerryneedell, 2019-01-11:
sure -- should SDA/SDC be the ones silkscreened because they definitely are not - they are SDA1/SCL1 in pins.c

jerryneedell, 2019-01-11:
Is there a schematic available?

ladyada, 2019-01-11:
@kattni can give you the schem (guide is coming up soon!)

kattni, 2019-01-11:
Provided!

dhalbert, 2019-01-11:
Tried an SHT31 with a recent CircuitPython build. Also doesn't work: it can't find the device. But it knows it's connected: it doesn't complain about "needs a pullup". I tried SCL1 and SDA1 (pins 24 and 25), and it works fine.

In CircuitPython, the pin mapping appears wrong: SCL and SDA are also 21 and 20, respectively, which are at the other end of that side of the board.

jerryneedell, 2019-01-11:
Ah -- OK - the LSM9DS1 also works using SDA1/SCL1 on 25/24 
It's not clear what is wrong with SCL/SDA on 21/20 since the schematic shows them going to the right pins - actually multiple pins...

jerryneedell, 2019-01-11:
looking at the arduino variants.h https://github.com/adafruit/ArduinoCore-samd/blob/master/variants/grand_central_m4/variant.h#L216
I am confused by the pin numbers -- how do these map to the schematic?

dhalbert, 2019-01-11:
ok, yes, pins 21 and SCL, and pins 20 and SDA are correspondingly connected together (verified with ohmmeter).

jerryneedell, 2019-01-11:
I verified that an MCP9808 sensor work using the SCL/SDA pins under Arduino

dhalbert, 2019-01-11:
Looks like some SERCOM errors in https://github.com/adafruit/samd-peripherals/blob/master/samd/samd51/pins.c. I wonder if there are differences between the SAMD51 chips. Anyway, diagnosed.

PR #1457 (2019-01-12): Use correct SERCOM for SDA/SCL on Grand Central
Incorrect SERCOM specified for SDA/SCL pins on Grand Central. Fixes #1456.
closed 2019-01-12 by tannewt