adafruit/circuitpython #1461: displayio - jpeg decoding

ladyada, 2019-01-13:
we have BMP support - JPEG would be used by a lot of people but needs to be in C for any reasonable speed/memory usage i think

here's a public domain decoder - https://github.com/Bodmer/JPEGDecoder

deshipu, 2019-01-14:
There is a PNG library in MicroPython: https://github.com/Ratfink/micropython-png

PTS93, 2020-08-15:
bitbank just released a really fast jpeg decoder that can target microcontrollers (and Linux)
https://github.com/bitbank2/JPEGDEC

Is there a preferred way how JPEG support should be implemented? As in the interface it should follow.
Copying the bitmap interface doesn't sound viable. You wouldn't want to construct a new JPEG on device f.e.

tannewt, 2020-08-17:
I think it's a bit tricky because CircuitPython isn't simply "decode image to screen". There is machinery around it that mixes pixels and tracks dirty areas.

There are a couple questions we'd need to answer:
1) Is it ok to store the decoded bitmap in memory?
2) Is it ok to store the JPEG in memory?
3) Could we cache decoded blocks in memory?

If 1, then simply decode into an existing Bitmap.
If 2, then decode the needed block from memory into memory when we get a pixel.
If 3, do 2 but store more than just the most recent block.
If we can't do 2 then the decode cost will be higher to get the encoded version from flash.

My guess is we want 3 or so which is like OnDiskBitmap but caches decoded block(s) in memory.

PTS93, 2020-08-17:
Not sure what you mean with caching decoded blocks, isn't that essentially the decoded bitmap?
My thinking so far was to do 1) and load JPEG from disk given that it is unlikely that M0+ could be supported with the already high memory requirements from CPY itself. Which pretty much leaves only MCUs which can easily hold a couple bitmaps in memory for even "higher res" 320x480 screens.

tannewt, 2020-08-18:
I wouldn't store the full decoded bitmap (aka all of the blocks) because it'll be one large allocation. If you store it at the decoded block level you have less memory overhead and more, smaller allocations. When evicting a block in favor of another the allocation can be reused then too.

bitbank2, 2020-08-19:
The code is very flexible about the output. It needs about 13.5K RAM for the internal tables, plus it defaults to a 4K output pixel buffer. This is to buffer as many MCUs as possible before sending to a display. You can tell it to only buffer one MCU at a time if you prefer. You can capture the pixels coming out and do with them as you please. It can generate RGB565 little or big-endian as well as 8-bit grayscale. If you have any questions, just ask.

PTS93, 2020-08-19:
> 
> 
> I wouldn't store the full decoded bitmap (aka all of the blocks) because it'll be one large allocation. If you store it at the decoded block level you have less memory overhead and more, smaller allocations. When evicting a block in favor of another the allocation can be reused then too.

ah ok thats fair. So you want to have the ability to use up possibly fragmented memory space by normally dealing with the blocks instead of a continuos bitmap block? Though is this really more memory friendly given that you have to keep track of all these blocks?

bitbank2, 2020-08-19:
Is it difficult to pass each block to the display rather than buffering them?

PTS93, 2020-08-19:
I would say it should be possible without violating the CPY philosophy. You would want to pass a displayio object to the JPEG library and have some functions in there to tell it where on the screen to draw the picture and possibly scale, color format etc.
The other use case would be with the GFX library where you might want to compose a framebuffer with the picture and not just draw it to the screen, I guess that might be the use case that Scott sees where you would have to handle it differently but I might misunderstand.

Though I think it should be ok to have both?

tannewt, 2020-08-19:
> ah ok thats fair. So you want to have the ability to use up possibly fragmented memory space by normally dealing with the blocks instead of a continuos bitmap block? Though is this really more memory friendly given that you have to keep track of all these blocks?

Right. It's easier to allocate small blocks into fragmented memory instead of small ones. I don't know how big each block would be but I'd do it at the JPEG unit level.

> Is it difficult to pass each block to the display rather than buffering them?

It's not easy because displayio is designed to mix different bitmap sources together before outputting to a display. That way it can minimize the bandwidth output to the display. It dictates the order of pixel access used to update the display.

> I would say it should be possible without violating the CPY philosophy. You would want to pass a displayio object to the JPEG library and have some functions in there to tell it where on the screen to draw the picture and possibly scale, color format etc.

Circumventing displayio to talk with the display directly would break down whenever you have something about the image like text because the composition is done MCU side, not on the display itself.

PTS93, 2020-08-19:
> Circumventing displayio to talk with the display directly would break down whenever you have something about the image like text because the composition is done MCU side, not on the display itself.

Sure! Thats why I thought it might be nice to have both ways. Either decode JPEG into bitmap and do composition with other elements or if you just want to draw it on the screen directly you can do that too by passing the decoder the displayio object to write to
For the case where f.e. you have drawn a UI and just want your JPEG to be rendered on top somewhere on that UI, this would be less resource hungry if you wanted to show multiple picture or switch between pictures quickly.

jepler, 2023-12-04:
https://www.atomic14.com/2023/09/30/a-faster-esp32-jpeg-decoder.html

tannewt, 2024-01-25:
@jepler want to close this?

closed 2024-01-26 by jepler

bitbank2, 2024-01-26:
FYI - I just optimized part of my JPEG decoder for the ESP32-S3's SIMD instructions. It gets a 20-40% speedup depending on the image and options. It's not an official release yet, but you can clone it and try.