adafruit/circuitpython #1481: Does not build for m4 metro express

mpeters2, 2019-01-21:
I get the following error when I try to build for metro_m4_express.  I'm using OSX and have been following these instructions: https://learn.adafruit.com/building-circuitpython/build-circuitpython

`~/circuitpython/ports/atmel-samd$ make BOARD=metro_m4_express V=1
GEN build-metro_m4_express/genhdr/mpversion.h
GEN build-metro_m4_express/genhdr/qstr.i.last
xargs: arm-none-eabi-gcc: No such file or directory
make: *** [build-metro_m4_express/genhdr/qstr.i.last] Error 127
make: *** Deleting file `build-metro_m4_express/genhdr/qstr.i.last'`

mpeters2, 2019-01-22:
Thanks for your help.  I downloaded and installed the gcc-arm stuff using these instructions (https://gnu-mcu-eclipse.github.io/toolchain/arm/install/), but I'm getting the same error.

I installed "arm-none-eabi-gcc" in /opt/gnu-mcu-eclipse.  Should it be somewhere else?

dhalbert, 2019-01-22:
@mpeters2 That is for use with the Eclipse; it is unfortunate that it shows up as a top Google result. For command line usage, just download 7-2018-q2- MacOS build https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads. Unpack it into some directory (doesn't have to be in /opt on any place where root is required to instsall it) and make sure that the executables are on your PATH.

closed 2019-01-23 by tannewt