adafruit/circuitpython #1483: Does not build for esp8266

larsks, 2019-01-22:
The current circuitpython master (9e8d182fe14c05b3e7c6a74e35987ec293310c7a) does not build for esp8266.  The build fails with:

```
../../py/objfun.c: In function 'fun_asm_call':
../../py/objfun.c:549:5: error: passing argument 2 of 'mp_arg_check_num' makes pointer from integer without a cast [-Werror]
     mp_arg_check_num(n_args, n_kw, self->n_args, self->n_args, false);
     ^
In file included from ../../py/objfun.c:33:0:
../../py/runtime.h:80:6: note: expected 'struct mp_map_t *' but argument is of type 'size_t'
 void mp_arg_check_num(size_t n_args, mp_map_t *kw_args, size_t n_args_min, size_t n_args_max, bool takes_kw);
      ^
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:55: build-feather_huzzah/py/objfun.o] Error 1
```

I'm building using:

```
$ xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6f) 4.8.5
```

kattni, 2019-01-22:
We've stopped supporting the ESP8266 in 4.0.0. That's why it's not building any longer.

closed 2019-01-22 by kattni

dhalbert, 2019-01-22:
We've dropped support due to ESP8266 not having native USB. We have written a library to support an ESP's (ESP8266 and ESP32) as a co-processor:
https://github.com/adafruit/Adafruit_CircuitPython_ESP_ATcontrol
and will be adding better support in the future (including hardware in this configuration).
Also see https://github.com/adafruit/Adafruit_CircuitPython_miniesptool

larsks, 2019-01-22:
Oh, sadness!  Thanks for the response.

ladyada, 2019-01-22:
@dhalbert lets remove the makefile?

Tasm-Devil, 2019-01-22:
A year ago, I was only a mouse click away from buying 28 Huzzah for my class.
Luckily I waited and bought the first six nrf52840 Express now. ;)
But sad for all esp owners, thats sure. I'm happy that you gave those boards a second chance.
The manufacturers of Octopus boards work around the corner. They are already experimenting with circuitpython.

ladyada, 2019-01-22:
Don't forget, you can still use Huzzahs with MicroPython! it works really well and MicroPython is doing an excellent job supporting the ESP8266 and ESP32 :)

Tasm-Devil, 2019-01-22:
Yea, I know. One student is still working with these Chips. He builds an RFid door lock for a workshop room and implements it with Micropython and ampy. He told me last week that he got quite familiar with the linux shell. That made me so proud XD