adafruit/circuitpython #1500: W5500 problem

notro, 2019-01-27:
I'm using an [Ethernet Shield for Arduino - W5500 Chipset](https://www.adafruit.com/product/2971) with an [Adafruit Grand Central M4 Express](https://www.adafruit.com/product/4064)

However I'm getting an error. Yesterday it always failed when sending, today it fails when receiving:
```
Adafruit CircuitPython patchbase-dirty on 2019-01-27; Adafruit Grand Central M4 Express with samd51p20
>>>
>>>
>>> import board
>>> import busio
>>> import wiznet
>>> import socket
>>> import time
>>>
>>> spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
>>> eth = wiznet.WIZNET5K(spi, board.D10, board.D11)
>>> eth.connected
True
>>> time.sleep(10)
>>> eth.ifconfig()
('192.168.10.135', '255.255.255.0', '192.168.10.1', '0.0.0.0')
>>>
>>> host = 'example.com'
>>>
>>> fam, typ, pro, nam, sockaddr = socket.getaddrinfo(host, 80)[0]
>>> ss = socket.socket(fam, typ, pro)
>>> sockaddr
('93.184.216.34', 80)
>>>
>>> ss.connect(sockaddr)
>>> ss.send(b"GET / HTTP/1.0\r\n\r\n")
18
>>> print(ss.recv(1024))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
>>>

```

It errors out in the [CHECK_SOCKMODE](http://elixir.tronnes.org/circuitpython/latest/ident/CHECK_SOCKMODE) macro:
```c
int32_t WIZCHIP_EXPORT(recv)(uint8_t sn, uint8_t * buf, uint16_t len)
{
   uint8_t  tmp = 0;
   uint16_t recvsize = 0;
   CHECK_SOCKNUM();
   CHECK_SOCKMODE(Sn_MR_TCP);
   CHECK_SOCKDATA();

```

This is what I did to make it build:
```diff
diff --git a/ports/atmel-samd/Makefile b/ports/atmel-samd/Makefile
index 797369bd6..55de847c0 100644
--- a/ports/atmel-samd/Makefile
+++ b/ports/atmel-samd/Makefile
@@ -281,6 +281,8 @@ SRC_C = \
        freetouch/adafruit_ptc.c \
        supervisor/shared/memory.c

+MICROPY_PY_NETWORK=1
+MICROPY_PY_WIZNET5K=5500
 ifeq ($(MICROPY_PY_NETWORK),1)
 CFLAGS += -DMICROPY_PY_NETWORK=1

diff --git a/ports/atmel-samd/mpconfigport.h b/ports/atmel-samd/mpconfigport.h
index da3383f8e..79e48312c 100644
--- a/ports/atmel-samd/mpconfigport.h
+++ b/ports/atmel-samd/mpconfigport.h
@@ -64,11 +64,11 @@
 #define MICROPY_STREAMS_NON_BLOCK   (1)

 #ifndef MICROPY_PY_NETWORK
-#define MICROPY_PY_NETWORK          (0)
+#define MICROPY_PY_NETWORK          (1)
 #endif

 #ifndef MICROPY_PY_WIZNET5K
-#define MICROPY_PY_WIZNET5K         (0)
+#define MICROPY_PY_WIZNET5K         (5500)
 #endif

 #ifndef MICROPY_PY_CC3K
diff --git a/shared-bindings/network/__init__.c b/shared-bindings/network/__init__.c
index 69f8bea60..4839e0cf5 100644
--- a/shared-bindings/network/__init__.c
+++ b/shared-bindings/network/__init__.c
@@ -28,6 +28,7 @@
 #include <stdint.h>
 #include <string.h>

+#include "py/obj.h"
 #include "py/objlist.h"
 #include "py/runtime.h"
 #include "py/mphal.h"
diff --git a/shared-bindings/socket/__init__.c b/shared-bindings/socket/__init__.c
index 29d47de56..5ecfa8d4d 100644
--- a/shared-bindings/socket/__init__.c
+++ b/shared-bindings/socket/__init__.c
@@ -28,6 +28,7 @@
 #include <stdio.h>
 #include <string.h>

+#include "py/obj.h"
 #include "py/objtuple.h"
 #include "py/objlist.h"
 #include "py/runtime.h"
diff --git a/shared-bindings/wiznet/__init__.c b/shared-bindings/wiznet/__init__.c
index e230deecc..fb8470462 100644
--- a/shared-bindings/wiznet/__init__.c
+++ b/shared-bindings/wiznet/__init__.c
@@ -28,6 +28,7 @@
 #include <stdint.h>
 #include <string.h>

+#include "py/obj.h"
 #include "py/objlist.h"
 #include "py/objproperty.h"
 #include "py/runtime.h"
diff --git a/shared-bindings/wiznet/wiznet5k.c b/shared-bindings/wiznet/wiznet5k.c
index 6d43e8992..288a19691 100644
--- a/shared-bindings/wiznet/wiznet5k.c
+++ b/shared-bindings/wiznet/wiznet5k.c
@@ -28,6 +28,7 @@
 #include <stdint.h>
 #include <string.h>

+#include "py/obj.h"
 #include "py/objlist.h"
 #include "py/objproperty.h"
 #include "py/runtime.h"
```

cc: @nickzoic

nickzoic, 2019-01-28:
Hmmm, okay, I don't have a Grand Central but I've got a Metro M4 Express which is hopefully similar enough to have the same problem, and a W5500 Feather Wing on an adaptor shield.  I'll have a look and check it's working for me ...

nickzoic, 2019-02-05:
OK I've got my devices set up running 4.0.0-beta.1-11-g388448215 and it's working for me 
with the Metro M4 Express and W5500 Feather Wing.

I did have some trouble when I first plugged it all together, with the Ethernet module on top the power is getting close to its limit and the board repeatedly reset itself until I plugged it into a better USB hub.
An external power supply should help there too.

If the remote server times out you'll get an Input/Output error as you described, though.

I'm a little puzzled by your build errors shown though: `make BOARD=grandcentral_m4_express
` is building fine here without modification.

notro, 2019-02-11:
I tried powering through the DC jack just in case I was balancing on the power limit, but it didn't help.
Pinging the board works just fine.

I think I'll just drop this for now. I was hoping for an easy way of getting network on this board so I could start porting the network modules in the CPython standard library. It will have to wait.

How do you enable the driver when building? AFAICT it's not enabled on any boards.

nickzoic, 2019-02-11:
It's enabled by default for some of the "express" boards but not the grand central for whatever reason.

Easiest way is to edit `ports/atmel-samd/boards/grandcentral_m4_express/mpconfigboard.mk` and add the lines:

```
MICROPY_PY_NETWORK = 1
MICROPY_PY_WIZNET5K = 5500
```

notro, 2019-02-14:
Thanks, I didn't check the makefiles...

I tried it on an Metro M4 Express and I get same result.
I can ping the board, it is able to resolve the dns name, but fails when sending:

```
Adafruit CircuitPython 4.0.0-beta.2 on 2019-02-05; Adafruit Metro M4 Express with samd51j19
>>>
>>>
>>> import board
>>> import busio
>>> import wiznet
>>> import socket
>>> import time
>>>
>>> spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
>>> eth = wiznet.WIZNET5K(spi, board.D10, board.D11)
>>> eth.connected
True
>>> time.sleep(10)
>>> eth.ifconfig()
('192.168.10.179', '255.255.255.0', '192.168.10.1', '0.0.0.0')
>>> host = 'example.com'
>>>
>>> fam, typ, pro, nam, sockaddr = socket.getaddrinfo(host, 80)[0]
>>> ss = socket.socket(fam, typ, pro)
>>> sockaddr
('93.184.216.34', 80)
>>> ss.connect(sockaddr)
>>> ss.send(b"GET / HTTP/1.0\r\n\r\n")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
>>>
```
The socked code works fine on an RPi sitting next to it.

notro, 2019-02-14:
I think 'll just close this for now and maybe pick this up later.

closed 2019-02-14 by notro

nickzoic, 2019-05-02:
@notro please see #1800 for what I hope is a fix!