adafruit/circuitpython #1501: __int__ not working

TG-Techie, 2019-01-27:
I am porting a friend's version of 2014 to cp and the code can't seem to use the __int__method:

here is a smaller but identical use case of what the problem is:

```
>>> 
>>> class x():
...     def __init__(self, value):
...         self.val = value
...         print(self.val)
...     
...     def __int__(self):
...         return self.val
...         
...         
... 
>>> x
<class 'x'>
>>> g = x(7)
7
>>> g.val
7
>>> g
<x object at 20002fd0>
>>> int(g)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert x to int
>>> bool(g)
True
>>> print(g)
<x object at 20002fd0>
>>> 
 ```

notro, 2019-01-27:
```__int__``` isn't supported.

This implements it:
```diff
 diff --git a/py/obj.c b/py/obj.c
index fb59eec82..8646f2d9d 100644
--- a/py/obj.c
+++ b/py/obj.c
@@ -242,7 +242,7 @@ bool mp_obj_equal(mp_obj_t o1, mp_obj_t o2) {
     return false;
 }

-mp_int_t mp_obj_get_int(mp_const_obj_t arg) {
+mp_int_t mp_obj_get_int(mp_obj_t arg) {
     // This function essentially performs implicit type conversion to int
     // Note that Python does NOT provide implicit type conversion from
     // float to int in the core expression language, try some_list[1.0].
@@ -255,6 +255,14 @@ mp_int_t mp_obj_get_int(mp_const_obj_t arg) {
     } else if (MP_OBJ_IS_TYPE(arg, &mp_type_int)) {
         return mp_obj_int_get_checked(arg);
     } else {
+#if MICROPY_CPYTHON_COMPAT
+        mp_obj_t dest[2];
+        mp_load_method_maybe(arg, MP_QSTR___int__, dest);
+        if (dest[0] != MP_OBJ_NULL && dest[1] != MP_OBJ_NULL) {
+            mp_obj_t meth = mp_obj_new_bound_meth(dest[0], dest[1]);
+            return mp_obj_get_int(mp_call_function_0(meth));
+        }
+#endif
         if (MICROPY_ERROR_REPORTING == MICROPY_ERROR_REPORTING_TERSE) {
             mp_raise_TypeError(translate("can't convert to int"));
         } else {
@@ -264,7 +272,7 @@ mp_int_t mp_obj_get_int(mp_const_obj_t arg) {
     }
 }

-mp_int_t mp_obj_get_int_truncated(mp_const_obj_t arg) {
+mp_int_t mp_obj_get_int_truncated(mp_obj_t arg) {
     if (MP_OBJ_IS_INT(arg)) {
         return mp_obj_int_get_truncated(arg);
     } else {
diff --git a/py/obj.h b/py/obj.h
index 74ce6a0a9..5be4245da 100644
--- a/py/obj.h
+++ b/py/obj.h
@@ -682,8 +682,8 @@ bool mp_obj_is_callable(mp_obj_t o_in);
 bool mp_obj_equal(mp_obj_t o1, mp_obj_t o2);

 static inline bool mp_obj_is_integer(mp_const_obj_t o) { return MP_OBJ_IS_INT(o) || MP_OBJ_IS_TYPE(o, &mp_type_bool); } // returns true if o is bool, small int or long int
-mp_int_t mp_obj_get_int(mp_const_obj_t arg);
-mp_int_t mp_obj_get_int_truncated(mp_const_obj_t arg);
+mp_int_t mp_obj_get_int(mp_obj_t arg);
+mp_int_t mp_obj_get_int_truncated(mp_obj_t arg);
 bool mp_obj_get_int_maybe(mp_const_obj_t arg, mp_int_t *value);
 #if MICROPY_PY_BUILTINS_FLOAT
 mp_float_t mp_obj_get_float(mp_obj_t self_in);
```

Maybe it would make sense to implement ```__float__``` and ```__index__``` as well.

I haven't got time to take this any further at the moment. Feel free to pick it up and make a PR.

microDev1, 2021-08-08:
Closing... `__int__` is working now.

closed 2021-08-08 by microDev1