adafruit/circuitpython #1504: no parallelio module for communicating with chips

TG-Techie, 2019-01-30:
I would love to talk to a display or parallel chips without (i haven't yet but will need to) bit banging the protocol. for example, the portal now is configured in parallel but I would love to have a more traditional library to talk to it in addition to displayio

tannewt, 2019-01-30:
What other chips are parallel? Could you give an example of the code you'd like to write?

TG-Techie, 2019-01-30:
my main interest in the ili9341 and other parallel displays

tannewt, 2019-02-01:
Ok, I've added the ability to have display bus (FourWire or ParallelBus) send a command. This is likely all you need. PR incoming with it.

TG-Techie, 2019-02-01:
awesome!  when working I'll post some code and close, sound good?

commit 448ae64d (2019-02-01): Add support for display rotation and raw commands

Display rotation is relative to the scan order of the display.
The scan order can be found by scrolling the display with command
0x37 `display_bus.send(0x37, struct.pack(">H", i % 128))`

Fixes #1504
closed 2019-02-01 by dhalbert