adafruit/circuitpython #1509: atmel-samd board port file system size

Sorunome, 2019-01-31:
So, I am trying to add a custom port for an atmel-samd based board (SAMD21), and a lot of work for that has already been done - great!

The problem I am facing is that, due to my larger bootloader (`0x4000` instead of `0x2000`) the compiled code doesn't fit. I also want to try to add in some other features which would also need up more flash.

So.....what better idea than to reduce the size of the flash filesystem! And...on we go down the rabbit hole. As it turns out, the flash filesystem size for samd21 boards being `0x10000` is hardcoded all over the place - i started hunting down the issue and got it to compile & run without corrupting itself. This means that the python REPL prompt via serial is working just fine - that's already good! However, the usb msd interface simply seems to be....not working. The device never shows up as a mass storage device on PC (linux) and I am a bit lost debugging here, so help would be appreciated.

Things I already had to change:
 * of course a custom `mpconfigboard.h` with different bootloader and flash sizes
 * set `CIRCUITPY_INTERNAL_NVM_SIZE` to `0` (why is this? other boards seem to do this even though they have internal storage?)
 * custom linker script with custom bootloader and flash size
 * add override in `mpconfigport.h` to **NOT** trigger `EXPRESS_BOARD`
 * make the hardcoded `0x10000` in `peripherals/samd/samd21/clocks.c` configurable

Am I missing anything else? (well, probably) Any pointers and/or help would be greatly appreciated

deshipu, 2019-01-31:
If the board project is not finalized yet, adding a flash chip for the filesystem to make it an express board would be probably the easiest way out of this.

Sorunome, 2019-01-31:
Thank you for the quick reply! Unfortunately it is already finalized, however it also has an SD card connected via SPI. Is it possible to use that as external flash (without breaking its FAT32 file system)? Thank you for answers!

tannewt, 2019-01-31:
Can you interact with the file system from the REPL? That's always my first step in debugging. After that I usually use a USB sniffer (a Beagle 12) to see how far MSD is getting. You can manage some of that from GDB as well.

I'm not sure what you asking about CIRCUITPY_INTERNAL_NVM_SIZE.

Sorunome, 2019-01-31:
> Can you interact with the file system from the REPL?

It appears to be blank, just nothing there. while `mkdir` and `chdir` don't error, `os.getcwd()` still just `"/"`, so probably borken

> I'm not sure what you asking about CIRCUITPY_INTERNAL_NVM_SIZE.

The name of that #define kinda implies to have to do something with the flash. Looking more into it, on [non-express-boards](https://github.com/adafruit/circuitpython) it seems to be set to `0`, and on [express-boards](https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/boards/metro_m4_express/mpconfigboard.h#L27) to some other number.

Looking more into it, it appears it is used [here](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/microcontroller/__init__.h#L51) and [here](https://github.com/adafruit/circuitpython/blob/master/shared-bindings/microcontroller/__init__.c#L179)

Thank you for the quick answer!

tannewt, 2019-01-31:
Ya, it sounds like the FS code itself is a problem. Once that works the USB usually does. I usually use GDB to see the output code of mkfs for insight.

The CIRCUITPY_INTERNAL_NVM_SIZE dictates how much memory is available via the `nvm` module for byte oriented storage rather that within the filesystem. We added it but don't know if it's used by anyone.

Sorunome, 2019-01-31:
It appears that FatFs can only format a file system if it has at least 50 blocks. Thank you for the help!

closed 2019-01-31 by Sorunome