adafruit/circuitpython #1514: hallowing does not boot with build 7c443fb.

jerryneedell, 2019-02-01:
I tried adafruit-circuitpython-hallowing_m0_express-en_US-20190131-7c443fb.uf2 and the Hallowing does not boot 
adafruit-circuitpython-hallowing_m0_express-en_US-20190131-63d4561.uf2  boots fine.

jerryneedell, 2019-02-01:
latest build also does not boot ;
https://adafruit-circuit-python.s3.amazonaws.com/bin/hallowing_m0_express/en_US/adafruit-circuitpython-hallowing_m0_express-en_US-20190201-7d039a5.uf2

same dmesg report
i

tannewt, 2019-02-01:
I think I've fixed this in #1513

jerryneedell, 2019-02-01:
yes you did ```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.1-38-g845783a45 on 2019-02-01; HalloWing M0 Express with samd21g18
>>> 

```

closed 2019-02-01 by jerryneedell