adafruit/circuitpython #1516: Supervisor problem

billr12, 2019-02-02:
supervisor.set_rgb_status_brightness(0) doesn’t take effect till after a ^c, preventing it from ever turning off the status led on a stand-alone device.  Tested on an ltsyBitsy M4 with CP 3.1.2

tannewt, 2019-02-04:
Can you please try it in the latest 4.0.0 Beta? Also, are you putting that in `code.py` or `boot.py`?

billr12, 2019-02-04:
Scott

I was putting the code in main.py.  When I moved it to boot.py, it worked perfectly.  I should have thought of that, particularly since I was already using boot.py to control logging.  Never gave it a thought!

I did try 4.0.0-beta.1 - works the same as 3.1

Thanks for your help,

Bill

tannewt, 2019-03-12:
I tried set_rgb_status_brightness with latest master on Grand Central from code.py and it worked just fine.

billr12, 2019-03-24:
Scott:

I just tried set_rgb_status_brightness on a PyPortal running beta 5.  With the command in boot.py, it works as expected.

With the command only in code.py (no boot.py) the led stays on when code is started either by powering the PyPortal or by hitting the reset button.  The led will be turned off by entering <ctrl>C in the Mu REPL while the code is running, or by re-saving the code while it is running.

I think this is the same as what we saw previously.

Bill

tannewt, 2019-03-26:
Ok, thanks for retesting @billr12. I'll reopen because this should be a quick fix. I have a feeling the brightness set isn't triggering an immediate write to the pixel when it should. A quick test would be to infinite loop after setting the brightness.

PR #1711 (2019-03-27): Make status LED brightness change immediate.
Fixes #1516
commit 503642e7 (2019-03-27): Make status LED brightness change immediate.

Fixes #1516
closed 2019-03-27 by dhalbert