adafruit/circuitpython #1517: Move nRF touch implementation to shared modules

tannewt, 2019-02-04:
It can be used generically.

nickzoic, 2019-02-05:
Yeah, there's no particular reason which makes this specific to nRF.

I'd also say if I was going to rewrite this I'd use a timer or interrupt handler on the falling edge of the pin instead of busy-waiting.  The current way is crude but effective :-/, given the microsecond timing involved not all bad but it'd be worth trying to be a little more clever ...

tannewt, 2019-02-05:
Crude but effective is also more portable. We currently don't have a common way of doing timers and interrupts across ports.

nickzoic, 2019-02-05:
> Crude but effective is also more portable. We currently don't have a> common way of doing timers and interrupts across ports.

Yeah, that's true: I went with the simplest possible approach since I
hadn't really looked at the GPIOTE functionality yet.
It might be worth developing some kind of abstraction though, in the
longer run ...

tannewt, 2019-02-06:
Yup, we can add an abstraction when it'd be useful.

ladyada, 2019-08-25:
@dhalbert this is done right?

dhalbert, 2019-08-25:
@ladyada Yes, it's been merged, available in the latest S3 builds for both nRF and SAMD51.

closed 2019-08-25 by ladyada