adafruit/circuitpython #1524: Samsung EVO Select 32GB does not initialize with CircuitPython 3.1.2 on Feather Express M4 with Adalogger

Helitronix, 2019-02-06:
First reported here with all the details: https://forums.adafruit.com/viewtopic.php?f=60&t=147212

Basic summary is that the Feather Express M4 / Adalogger hardware works with other micro SD cards that I have now tried, but the Samsung EVO 32GB card fails to initialize with CircuitPython 3.1.2 and the February 4'2019 libraries.  The card is properly initialized if I use the SD Cardinfo sample application in Arduino mode.  So the hardware is working fine, but something isn't quite right in the CircuitPython stack with this card.

danhalbert asked that I submit the bug here.  Thanks for your help!

kattni, 2019-02-06:
I was unable to get the same SD card to initialise on the Grand Central running 4.x.alpha. I attributed it to the size at the time, however I don't have any smaller EVO SD cards to test, so I can't be certain. I was able to get an off-brand 8GB SD card to work successfully.

makermelissa, 2019-02-06:
I should be getting my Grand Central in the next couple hours and I have some 32GB Samsung EVO cards that I could try.

makermelissa, 2019-02-06:
Yeah, I had the same issue on a Grand Central running Beta 2 with the 32GB SD card. I'll have to pull out my collection and try various ones.

Helitronix, 2019-02-06:
@kattni 
@makermelissa 

Thanks for your comments.  I have a Transcend 8GB, Sandisk 4GB and Kingston 2GB that do work properly.  Of course, the first card I tried was the new 32GB Samsung that didn't work.  :-) I'm glad I found something that works, but it sure would be better if these Samsung cards worked, too.

dhalbert, 2019-02-06:
Note this previous issue with 32GB Samsung EVO cards, which was fixed on M0. 

https://github.com/adafruit/circuitpython/issues/245

Probably worth retesting on an M0 to confirm it’s chip-specific.

makermelissa, 2019-02-06:
Ok, I hooked up an SD Card breakout (I don't have the adalogger) to both a Feather M4 Express and a Feather M0 Express and get the following error on both:
```
OSError: timeout waiting for v2 card
```