adafruit/circuitpython #1533: busio.SPI.write_readinto sometimes returns one less byte

anandjog, 2019-02-11:
In some SPI slaves, a read from a register is achieved by writing to it.  I've been using the busio.SPI.write_readinto() function described at

https://circuitpython.readthedocs.io/en/2.x/shared-bindings/busio/SPI.html

In order to read the contents of a register, I need to send the 8 bit register address + read bit (all included in the first 8 bits) and an additional dummy byte.  The contents of the register get clocked out during the dummy byte write.

Although I specify 2 byte write and read buffers, CircuitPython sometimes returns only one byte.  Although this is the correct byte (with the data contents of the register) there could be issues for users who rely on multi-byte burst reads
[image]

dhalbert, 2019-02-11:
Could you specify the board you are using and the version of CircuitPython?

anandjog, 2019-02-11:
I'm using CircuitPython 3.1.2 (the latest stable release) on the Metro M0 express.  Also seeing spotty behavior with certain data writes (e.g 0x55) and read back..  The chip AMIS-30532 has been extensively used with other microcontrollers...

anandjog, 2019-02-11:
Here's my CircuitPython code if it helps

# source code for metro m0 express
import busio
import board
import digitalio
import supervisor

baudrate = 100000
# the following is for SPI communications
cs = digitalio.DigitalInOut(board.A2)   # cs should be wired to this pin
cs.direction = digitalio.Direction.OUTPUT
cs.value = True # chip select default state is HIGH
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)

def BytesToIntsList(writeDataBytes):
    # writeDataBytes = ['0A', '75', 'C3', '05'] every byte must be two hex characters e.g 0x01 and not 0x1
    intlist = [int(hexbytestr, 16) for hexbytestr in writeDataBytes]
    return intlist

def spi_read(regAddress, nbytes):
    # spi read from a register requires nbytes dummy writes to the same register

    data_out = bytearray([int(regAddress, 16)] + nbytes*[0])
    result = bytearray(nbytes+1)

    while not spi.try_lock():
        pass

    # read portion
    spi.configure(baudrate=baudrate, phase=0, polarity=0)
    cs.value = False
    spi.write_readinto(data_out, result)
    cs.value = True
    spi.unlock()
    return result

def spi_write(writeDataList):
    """
    :param writeDataList: - a list of integers (0-255 each)
    :return:
    """
    while not spi.try_lock():
        pass

    spi.configure(baudrate=baudrate, phase=0, polarity=0)
    cs.value = False
    spi.write(bytes(writeDataList))
    cs.value = True
    spi.unlock()
    return

while True:
    if(supervisor.runtime.serial_bytes_available):
        inbytes = input()     # calling code must send an '\r' terminated utf-8 byte sequence
        instr = str(inbytes, 'utf-8')
        cmd_tokens = instr.split('_')
        if cmd_tokens[0] == 'SPI':  # user has requested an SPI read or write
            regAddress = cmd_tokens[2]
            if cmd_tokens[1] == 'R':
                nbytes = int(cmd_tokens[3])
                read_result = spi_read(regAddress, nbytes=nbytes)
                print(read_result)
            if cmd_tokens[1] == 'W':
                writeDataBytes = cmd_tokens[3:]   # a string of octets
                writeDataList = BytesToIntsList([regAddress] + writeDataBytes) # concatenate bytes
                print('writing %r' % writeDataList)
                spi_write(writeDataList=writeDataList)

dhalbert, 2019-02-12:
>Although I specify 2 byte write and read buffers, CircuitPython sometimes returns only one byte.

I'm not sure what you mean by this. You create the read buffer, and it's 2 bytes long, based on what you said. Do you mean the read buffer is only one byte long when you return it, or do you mean there's only one byte of meaningful data in it? That would be true because the register address gets transmitted during the first byte time.

Do you have some sample output or something like that we could look at?

dhalbert, 2019-02-12:
I think you could just do
```
cs.value = False
spi.write(<1 byte buffer of register address>)
spi.readinto(<1 byte buffer of register data>)
cs.value = True
```
`spi.readinto()` writes out zeros as it's reading in, which is the same thing you're doing by using `write_readinto()`, based on the diagram above.

There are SPI devices that let you pipeline the next command as the data requested in the previous command is coming back. That's why we added `write_readinto()`.

anandjog, 2019-02-12:
Hello @dhalbert 
I think the main problem lies with the bytearray function in CircuitPython or my misunderstanding of it.

Suppose I want to write the hex value '55' (0x55) to a register (0x03).  While writing these bytes we end up generating a list of integers [03, 85] where the first integer is the register address and the second is the integer equivalent of '0x55' = 85.

When we read back, the bytearray function represents the '0x55' as 'Latin capital letter U' per the following link

https://www.utf8-chartable.de/unicode-utf8-table.pl

The CircuitPython program at the print(read_result) statement (line 65) of my code should output bytes that represent hex.  So the return string (sent over the serial port by CircuitPython)

bytearray(b'UU') 

should actually read

bytearray(b'\x55\x55'\r\n') for me to be able to correctly parse.  

This behavior is number-dependent.  So for 0xAA there are no issues.  But for 0x55, it outputs 'U' or latin upper case U.

Program output is attached with comments for your convenience.  Many thanks!

Anand
[utf_latin_decode_bug.txt](https://github.com/adafruit/circuitpython/files/2853618/utf_latin_decode_bug.txt)

anandjog, 2019-02-12:
@dhalbert 

Its the print(read_result) statement in the CircuitPython code.  

Consider 
a = bytearray([85, 85])
print(a)       # this outputs bytearray(b'UU')

a = bytearray([170, 170])
print(a)       # this outputs bytearray(b'\xaa\xaa')

I'm wondering if there is an option in the print statement that will do the correct conversion

dhalbert, 2019-02-12:
So the problem is that `print(read_result)` is printing the wrong thing, is that correct?

I think what you want to do something like `print(hex(read_result[1])[2:])`

That will retrieve the second byte of the bytearray that read_result returns, convert it to hex (like `'0x55'`) and then remove the `'0x'`  (using `[2:]`).

I think you may know this already, but subscripting a bytearray returns an integer, not a single byte bytestring. This is unlike strings: if you subscript a string, you get a one-character string.

I am not exactly sure from your debugging output what you are sending and expecting back. But `hex()` will convert an integer to a hex value (if that's what you want). If you just want the integer value, then just do `read_result[1]`, which will return an integer with the register value.

dhalbert, 2019-02-12:
Printing a bytearray sometimes prints characters and sometimes prints escaped hex values (if there's no equivalent printing character). That's confusing, I agree.

anandjog, 2019-02-12:
@dhalbert 

You have hit the nail on the head!  

Yes, print(read_result) is printing the wrong thing - actually I'm misusing it.  

Since I want my code to work with burst reads I need to generalize it for multiple bytes.  I'll work on this a bit more but at least we know it was my mistake :-)

Thanks for being there for me!

Anand

dhalbert, 2019-02-12:
Happy to help!

closed 2019-02-12 by dhalbert