adafruit/circuitpython #1539: DISPIO: dispio doesn't like try excepts??

TG-Techie, 2019-02-13:
in this and another program, it just crashes after going through a try except

```
import TG_Display_Emulator as disp, board, time

disp.DISPLAY = board.DISPLAY
disp.WIDTH = 320
disp.HEIGHT = 240

for i in range(255):
    try:
        disp.rect(i,i,i,i,disp.color(255-i, i//2 * 128, i))
    except:
        pass
```
crashes the pyportal

need:
https://github.com/TG-Techie/TG_CircuitPython_Display_Emulator

tannewt, 2019-02-13:
@TG-Techie Do you have a debugger handy like a JLink? That would allow you to get a backtrace.

TG-Techie, 2019-02-13:
not hand, I'm back near one on Saturday after 5

tannewt, 2019-02-14:
Please try it with #1548.

Smaller examples that cause the crash are handier as well. Thanks!

tannewt, 2019-02-15:
@TG-Techie did you get a chance to test this with the latest code?

tannewt, 2019-02-20:
@TG-Techie Please reopen if this is an issue still.

closed 2019-02-20 by tannewt