adafruit/circuitpython #1545: nrf52840 crashes when importing featherwing library

jerryneedell, 2019-02-13:
Executing this line from the REPL on an nrf52840 succeeds but if it is placed ina script and executed the board crashes - drops USB connection and enters safe mode
```from adafruit_featherwing import alphanum_featherwing```

jerryneedell, 2019-02-14:
to clarify -- I test this as follows:
create a script named alnum.py containing the single line 
```from adafruit_featherwing import alphanum_featherwing```

then copy alnum.py to the CIRCUITOY drive and execute it via the REPL:
```import alnum```
the board immediately disconnects from USB and the NEOPIXEL goes Yellow indicating safe mode.

pressing RESET brings it back to normal

If you copy/paste the same command to the REPL it executes normally.

jerryneedell, 2019-02-14:
hmmm -- I hope this is a clue: If I cop yht contents of alphanum_featherwing.py
```
import adafruit_ht16k33.segments as segments
from adafruit_featherwing import shared
from adafruit_featherwing.led_segments import Segments

class AlphaNumFeatherWing(Segments):
    """Class representing an `Adafruit 14-segment AlphaNumeric FeatherWing
       <https://www.adafruit.com/product/3139>`_.

       Automatically uses the feather's I2C bus."""
    def __init__(self, address=0x70):
        super().__init__()
        self._segments = segments.Seg14x4(shared.I2C_BUS, address)
        self._segments.auto_write = False
```

directly into the featherwing_alphanum_simpletest.py replacing
```
from adafruit_featherwing import alphanum_featherwing
```
and changing
```
display = alphanum_featherwing.AlphaNumFeatherWing()
```
to
```
display = AlphaNumFeatherWing()
```

then the demo runs fine on my feather_nrf52840_express

I hope this makes sense to someone ;-)

jerryneedell, 2019-02-14:
Could this issue have something to do with the execution of multiple layers of imports ? Is that different on the atmel_samd51 from the nrf_52840?

jerryneedell, 2019-02-15:
I have simplified the test to cause this issue:
create two scripts
test.py containing this line
```
import adafruit_ht16k33.segments as segments
```
and test1.py containing:
```
import test
```
copy both files onto the feather_nrf52840

at the REPL
```
import test1
```
the nrf52840 crashes -- drops USB and enter safe mode

note: running import test  at the REPL works...

jerryneedell, 2019-02-15:
I alos tried importing from a different library 
in  test.py:
```
import adafruit_bus_device.i2c_device as myimport
```

this does not cause the crash.  Perhaps it is something in the ht16k33 lib

jerryneedell, 2019-02-15:
forgot to post safe mode message 
```

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-58-g35e3d99f6 on 2019-02-14; Adafruit Feather nRF52840 Express with nRF52840
```

jerryneedell, 2019-02-15:
Ozone screensho after trap to HardDault_Handler  -- unsupported return stack adderss -- not much information
[image]

jerryneedell, 2019-02-15:
gdb backtrace:
```
(gdb) break HardFault_Handler
Breakpoint 1 at 0x44686: file supervisor/port.c, line 121.
(gdb) c
Continuing.

Breakpoint 1, HardFault_Handler () at supervisor/port.c:121
121	    reset_into_safe_mode(HARD_CRASH);
(gdb) tr
Note: breakpoint 1 also set at pc 0x44686.
Tracepoint 2 at 0x44686: file supervisor/port.c, line 121.
(gdb) his
Undefined command: "his".  Try "help".
(gdb) backtr
#0  HardFault_Handler () at supervisor/port.c:121
#1  <signal handler called>
#2  qstr_data (q=q@entry=65, len=len@entry=0x2003ec3c) at ../../py/qstr.c:267
#3  0x0003affa in mp_obj_str_get_data (self_in=0x106, len=len@entry=0x2003ec54)
    at ../../py/objstr.c:2148
#4  0x0003f300 in find_file (file_str=file_str@entry=0x2003e652 "adafruit_bus_device", 
    file_len=file_len@entry=19, dest=0x2003ee8c, dest@entry=0x2003eeb4)
    at ../../py/builtinimport.c:124
#5  0x0003f68a in mp_builtin___import__ (n_args=n_args@entry=5, args=args@entry=0x2003eecc)
    at ../../py/builtinimport.c:393
#6  0x00032416 in mp_import_name (name=<optimized out>, fromlist=<optimized out>, 
    level=<optimized out>) at ../../py/runtime.c:1417
#7  0x000414b8 in mp_execute_bytecode (code_state=code_state@entry=0x2003ef60, 
    inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1217
#8  0x00036f74 in fun_bc_call (self_in=0x20008a20, n_args=0, n_kw=206523, args=0x0)
    at ../../py/objfun.c:284
#9  0x000326ba in mp_call_function_n_kw (fun_in=fun_in@entry=0x20008a20, n_args=n_args@entry=0, 
    n_kw=n_kw@entry=0, args=args@entry=0x0) at ../../py/runtime.c:649
#10 0x000326e4 in mp_call_function_0 (fun=fun@entry=0x20008a20) at ../../py/runtime.c:623
#11 0x00032732 in mp_parse_compile_execute (lex=lex@entry=0x20008a20, 
    parse_input_kind=parse_input_kind@entry=MP_PARSE_FILE_INPUT, 
    globals=globals@entry=0x2003e6b0, locals=locals@entry=0x2003e6b0) at ../../py/runtime.c:1513
#12 0x0003f3d0 in do_load_from_lexer (module_obj=module_obj@entry=0x2003e6c0, lex=0x20008a20)
    at ../../py/builtinimport.c:152
#13 0x0003f41a in do_load (module_obj=module_obj@entry=0x2003e6c0, file=0x2003f26c, 
    file@entry=0x2003f294) at ../../py/builtinimport.c:245
---Type <return> to continue, or q <return> to quit---
```

makermelissa, 2019-02-15:
I've been working on approaching this a slightly different way. I've been removing a lot of contents of the libraries that import it and it seems to have something to do with an import chain. I'm not sure why it's specific to the nRF52840, but maybe Jerry's comments would help shed more light on that. Either way, I was also able to reproduce this bug.

jerryneedell, 2019-02-15:
Just a wild guess, but I wonder if it is something to do with running out of stack space.. I found that if I compile with DEBUG=1 I can reproduce the problem -- see above, but when I tried to run some tests of the ESP32SPI code, It also went to Safe Mode but in that case, the Safe Mode message reported the reason as running out of stack space.  I normal build w/o DEBUG runs the ESP32SPI code fine on the nrf52840.

makermelissa, 2019-02-15:
Yeah, that makes sense.

makermelissa, 2019-02-15:
Fortunately running with code.py works just fine and I think that's how most users would approach it.

jerryneedell, 2019-02-15:
oh - that is interesting -- I never tried that -- yuk :-(

dhalbert, 2019-02-15:
>Just a wild guess, but I wonder if it is something to do with running out of stack space..

I just noticed (while doing something else) that the default nRF stack is only 4k. I am adjusting that. It would be interesting to know if this also fails on an M0, where the default stack is also that size.

jerryneedell, 2019-02-15:
yes -- raising it to 8K fixes the problem!!

jerryneedell, 2019-02-15:
Increasing the stack size to 8K resolved this issue.  Tested on feather_nrf52840_express and particle_argon.  

I guess we should leave this issue open until the nrf52840 builds are reconfigured for larger stack -- @dhalbert has a PR in work for that and other changes.

tannewt, 2019-02-15:
It would be nice to add stack checking to throw an exception in this case. Raising the stack size isn't a great fix because other imports can still cause the same crash.

jerryneedell, 2019-02-16:
fixed by #1554

closed 2019-02-16 by jerryneedell