adafruit/circuitpython #1549: DISPIO: zero Shape width crashes and bootloops cp

TG-Techie, 2019-02-14:
try making s shape have no width! then putting it in a group and showing it

it should raise a ValueError: Width of Shape cannot be zero

TG-Techie, 2019-02-14:
height too

ladyada, 2019-02-14:
@TG-Techie if you can write a minimal (but complete) example code for these bugs to demonstrate it crashing, plus the output of the crash that will really help us find and squish these bugs! the more details/info, the better as it lets us focus on fixing :)

TG-Techie, 2019-02-14:
Totally will write one up now
error message: starts boot looping and does not appear on usb 

```
from displayio import *

shp = Shape(20, 0)
```

tannewt, 2019-02-14:
Please update to the latest master. I fixed this with #1548. Thank you for the small code example.

Also, it should no longer boot loop. It should enter safe mode now after a crash.

closed 2019-02-14 by tannewt