adafruit/circuitpython #1556: bleio: cccd not properly after Peripheral is added to a Service?

edwios, 2019-02-16:
Hi,

I have the following piece of code, however when I run it, it returned CCCD error:
OSError: Failed to read CCCD value, err 0x3002                                  

```
	chara = bleio.Characteristic(uuid_char_temp, notify=True, read=True, write=False)
	serv = bleio.Service(uuid_env_sense, [chara])
	periph = bleio.Peripheral([serv], name="BME680")

	periph.start_advertising()

	while True:
		send_temp()
		time.sleep(5)
```

`send_temp()` is essentially `chara.value = bytearray([temp & 0xFF, temp >> 8])`

Did I missed anything?

Note: Circuitpython master tree on Feb 8th, nRF52840 dongle

Regards,
Ed

edwios, 2019-02-16:
Found what's wrong, it was due to a bug causing `send_temp()` to set the `chara.value` before `periph.connected` returned True.

Closing the issue.

BTW, great works on the **bleio**, it is really friendly and very easy to use!

closed 2019-02-16 by edwios

dhalbert, 2019-02-16:
One question is whether it should be safe to set the value when not connected. Maybe it should just suppress trying to notify in that case, since you may want to set an initial value.

Thanks for the compliment! Getting to a pleasant API is hard.

edwios, 2019-02-24:
For a peripheral, I don't know what is the point of writing to a characteristic when it is not connected.

dhalbert, 2019-02-24:
You might want to initialize the characteristic with a value before you start advertising. When a client connects, they might request the value immediately.

tcpipchip, 2019-08-03:
right!