adafruit/circuitpython #1561: RTC Functionality missing from NRF making DS3231 Library Unusable

makermelissa, 2019-02-17:
Regarding Discussion in PR https://github.com/adafruit/Adafruit_CircuitPython_FeatherWing/pull/27, in the latest official release (beta 2), the functions are there, but cause a BCDDateTimeRegister object to be returned. @jerryneedell had compiled a build from master on 2/15 and that was missing the mktime() and localtime() functions, but the BCDDateTimeRegister seemed to be fixed.

makermelissa, 2019-02-17:
Forgot to mention, Dan mentioned there was work being done on this in #1534.

makermelissa, 2019-02-17:
Ok, now that I've successfully compiled the latest version of circuitpython, it appears the specific functionality I need is the mktime() and localtime() functions in the time package.

nickzoic, 2019-04-02:
G'day Melissa!  PR #1534 did not turn out as well as I'd hoped but hopefully the PR #1736 should address this ...

nickzoic, 2019-04-09:
G'day @makermelissa #1736 is merged & I think it works for my very small test case, can you confirm that this works okay for you now?

makermelissa, 2019-04-09:
Tested and it works great! Thanks @nickzoic.

closed 2019-04-09 by makermelissa