adafruit/circuitpython #1565: trinkit analogout fails after soft reset

jerryneedell, 2019-02-18:
The following test script will execute after a hard RESET, but fails with an I/O Error after a soft reboot

test.py
```
from analogio import *
from board import *
aout = AnalogOut(D1)
```

```
HARD RESET
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-79-gfd3e9ce2f on 2019-02-17; Adafruit Trinket M0 with samd21e18
>>> import test
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-79-gfd3e9ce2f on 2019-02-17; Adafruit Trinket M0 with samd21e18
>>> import test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 3, in <module>
OSError: [Errno 5] Input/output error
>>> 

```

ladyada, 2019-02-18:
are you saving with Mu?

makermelissa, 2019-02-18:
Jerry, I and iraytrace (on discord) all experienced this with a fresh build off master on CP. I personally saved using Mu, and I think the others did as well.

jerryneedell, 2019-02-18:
I was not using Mu. Just running from REPL And typing control-d to reboot. I was not using code.py. Just test.py.

ladyada, 2019-02-18:
ok yeah i have seen this lately but i thought it was related to save failing. i noticed that if i opened code.py in Mu and saved it (maybe changing a char and deleting) then it would magically work, or maybe its unreleated

PR #1573 (2019-02-20): Fix #ifdefs in port.c to call xxx_reset() routines properly. Remove most uses of EXPRESS_BOARD.
Fixes #1565.

`supervisor/port.c` was using `EXPRESS_BOARD`, which had been removed from the `Makefile`, so those `#ifdefs` were not working properly, and some `xxx_reset()` routines were not getting called on soft restart. Changed to use `INTERNAL_FLASH_FILESYSTEM` and CIRCUITPY_xxx` instead

But `INTERNAL_FLASH_FILESYSTEM` which was defined as a make variable but not a preprocessor macro, so added that. `QSPI_FLASH_FILESYSTEM` and `SPI_FLASH_FILESYSTEM` were already macros. Now we actually check t [... cut]
closed 2019-02-20 by tannewt