adafruit/circuitpython #1567: DISPIO: changing tilegrid breaks containing group

TG-Techie, 2019-02-18:
```
from board import DISPLAY
import displayio, time
class rect(displayio.TileGrid):

    def __init__(self, x, y, width, height, color):

        self.plt = displayio.Palette(2)
        self.plt.make_transparent(0)
        self.color = color

        super().__init__(displayio.Shape(width, height), pixel_shader = self.plt, position = (x,y))

    @property
    def color(self):
        return self.plt[1]

    @color.setter
    def color(self, valin):
        self.plt[1] = valin

    @property
    def x(self):
        return self.position[0]

    @property
    def y(self):
        return self.position[1]

    @x.setter
    def x(self, valin):
        self.position = (valin, self.position[1])

    @x.setter
    def y(self, valin):
        self.position = (self.position[0], valin)

grp = displayio.Group(max_size = 1)
r = rect(5,5,50,30,0xffffff)
print(type(r))
grp.append(r)
print(type(r))
DISPLAY.show(grp)
r.x = 50
time.sleep(1)
print(type(grp[0]))
```
causes:
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
<class 'rect'>
<class 'rect'>
Traceback (most recent call last):
  File "main.py", line 47, in <module>
NameError: name 'grp' is not defined

TG-Techie, 2019-02-18:
note: removing the r.x = 50 makes it run perfectly

tannewt, 2019-03-12:
I think this is due to the position setter not handling subclasses correctly. Fix incoming.

commit 224e9b10 (2019-03-13): Standardize TileGrid to x and y properties over position

This brings it inline with Group. Also fixes #1613

This also includes a number of fixes for where a method is called
through a subclass. We now correctly get the native object.
Fixes #1567

Lastly, this adds subscript support to TileGrid for changing tile
indices. Similar to Bitmap, it accepts ints or 2-tuples.
closed 2019-03-13 by dhalbert