adafruit/circuitpython #1570: PYPORTAL: collect inconsistently causing softreload!!!!!!!!!

TG-Techie, 2019-02-18:
I will attach the files but I cannot recreate the issue consistently. also the code is very complex 
every now and then when it tries to collect it reloads
```
ReloadException: 
soft reboot
```

change the pyportal to spi mode via the jumper and run these files
[portal backup.zip](https://github.com/adafruit/circuitpython/files/2877301/portal.backup.zip)

ladyada, 2019-02-18:
TG, i edited to remove the exclaimation points. please keep the issue text and subject lines informative and compact

TG-Techie, 2019-02-18:
my apologies, I should have proofread the issue title.

tannewt, 2019-02-20:
Is the device connected to your computer via USB? Any write activity your host does will cause a reload.

TG-Techie, 2019-02-21:
@tannewt  yes it is plugged in but i was not saving. i would ctrl-d to make it reload then as it was running it would reload on it's own.

tannewt, 2019-02-21:
Your OS may be doing stuff on the drive in the background. If you see reload errors when usb isn't connected please reopen. Otherwise, this is as expected.

closed 2019-02-21 by tannewt