adafruit/circuitpython #1574: I2C scan/address bit order error

alexwhittemore, 2019-02-19:
It sure looks a lot like there's a bit order reversal when handling i2c addresses.

I've got a custom board - call it a Trinket M0 (closest relative). I'm using an i2c port on SDA=PA22 and SCL=PA23. This port seems to be working normally, but only if I bit-reverse addresses. For instance, I've got a fuel gauge attached to the port with an address of 0x6C. However, if I i2c.scan():
```
>>> hex(i2c.scan()[0])
'0x36'
```

I can carry on using the device as expected, as long as I continue using this reversed address of 0x36.

What's especially odd is that a logic analyzer trace of the transactions with the reversed address look correct, for instance
```
i2c.writeto(0x36, bytes([0x0A]), stop=False)
result = bytearray(2)
i2c.readfrom_into(0x36, result)
return result
```
gets me
[image]

Note that the address (reversed in python) comes through correctly, but so does the 0x0A byte, which remains in-order as-coded.

An aside, it looks like the subsequent read has the wrong address, but I think this is a Saleae bug - they read the LSB to determine read/write, but then don't right-shift the word before decoding the address, so a trailing 1 indicating a read gets included in the decode. I checked the bits on rising edges manually and they line up.

Anyway I'll try this on a proper M0 as well to see if it's maybe something to do with the logic in handling the alternate SERCOM or something.

ladyada, 2019-02-19:
hi- we use 7-bit addressing, not 8-bit 
 0x6C / 2 = 0x36 :)

closed 2019-02-19 by ladyada

alexwhittemore, 2019-02-19:
Ahhh, so "8 bit addressing" doesn't mean "some clever way of adding an extra bit" - it means "we incorrectly left-shift the _true_ device address when we really shouldn't"

Clear enough, I suppose!

ladyada, 2019-02-20:
hmm, not sure your question but i think your datasheet is probably confusing, and has the '8 bit' address which includes a 0 for the lowest bit (left shift one). we use the 'raw' address, same as Arduino and Linux, which does not pad the LSB
https://www.i2c-bus.org/addressing/
maybe that helps?

alexwhittemore, 2019-02-20:
Yes - that's exactly it. I never realized the difference. Thanks for walking me through it!