adafruit/circuitpython #1575: Problem installing 4.0 beta1 & beta2 on NRF52840 Feather Express

ATMakersBill, 2019-02-19:
Hi folks, I have a Feather NRF52840 and tried to install CP 4.0b2 on it.  I entered bootloader mode and got the FTHR840BOOT drive.  I copied the UF2 and got a failure
[image]

I have gotten different errors saying that the file is too large to fit on the device, but it always fails.

Here is my bootloader version.
```
UF2 Bootloader 0.2.6 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Dec 21 2018
```
After multiple attempts, I was able to get the REPL started on beta1.  I tried erase_filesystem() and it made no difference.

I switched to a second device and started with just beta1.  That gave me an odd error saying the file had not been copied... but it then started up CIRCUITPY and works.

I still have the broken device (which is still broken) if you want any testing done.

This has been replicated on Discord

dhalbert, 2019-02-20:
This is on Windows 10, 1809 build. I got a different error but also failed to copy the .uf2 to the Feather 52840.
```
UF2 Bootloader 0.2.6-10-g641d9fe lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Jan 31 2019
```

[image]

Upgraded to bootloader 0.2.8, though it reports as:
```
UF2 Bootloader 0.2.6-11-gd9b275c lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Feb 19 2019
```
Then I get the error shown in the original post, where the UF2 is reported to have disappeared from its original location.

dhalbert, 2019-02-20:
On Windows 7, trying to copy via drag and drop shows this error:

[image]

dhalbert, 2019-02-20:
Re-created as https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/46, closed here, since it's a bootloader issue.

closed 2019-02-20 by dhalbert