adafruit/circuitpython #1579: PYPORTAL

TG-Techie, 2019-02-21:
Since i upgraded to the most recent master,4.0.0-beta.2-97-g0dc260058 on 2019-02-20
, occasionally while editing with mu the pytportal freezes up and the led goes red and the mounted drive will freeze up and crash too if you try to interact with it. then when unplugged and re-plugged some file on the drive will be corrupted, or an entire folder. i have a jtag edu with me (the large one) if you think that could help. this is the only board this happens on that i have. i wish i had more data to report

ladyada, 2019-02-21:
this this a duplicate of https://github.com/adafruit/circuitpython/issues/1577#issuecomment-465834205 ?

tannewt, 2019-02-21:
Let's focus on #1577. These are similar enough that we only need one.

closed 2019-02-21 by tannewt