adafruit/circuitpython #1590: BLEIO Scanner incorrect addresses and other issues

rdagger, 2019-02-23:
I'm trying to use a Feather nRF52840 Express to detect BLE beacons. I'm using the following sample code:
```
import bleio
scanner = bleio.Scanner()
entries = scanner.scan(2500)
```
I've notices 4 issues.

1. Printing entries shows the same hardware address and raw data for all entries.  The RSSI for the entries is different.
2. The 4th hex value of the address is incorrect.  It is showing the 5th hex value in both the 4th and 5th positions (Example: AA:BB:CC:**EE**:EE:FF instead of AA:BB:CC:**DD**:EE:FF).
3. The entry.address property doesn't work (TypeError: extra keyword arguments given).
4. The scanner.scan method has to be run twice to return any entries.

dhalbert, 2019-02-23:
Thanks for noticing the issues. The `Scanner` code hasn't been worked on at all, and is from an earlier PR. We're not doing central mode for 4.0.0. We should probably remove these from the build right now. I'll keep this open as reference for when we do work on it.

rdagger, 2019-02-23:
Thanks for the quick response.  I'm interested in detecting BLE beacons by address and signal strength.  Is that going to be possible?  Are there any current workarounds?

dhalbert, 2019-02-23:
We have plans for central mode, which will include the kind of scanning you want, post 4.0.0. We're trying to get 4.0.0 out soon (no published schedule). If time is pressing, our nRF52 Arduino board support package has scanning, for example: https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Central/central_scan/central_scan.ino

dhalbert, 2019-02-24:
Superseded by #1595.

closed 2019-02-24 by dhalbert