adafruit/circuitpython #1595: Implement BLE Scanner

dhalbert, 2019-02-24:
... or the equivalent, so that beacons can be detected.

rdagger, 2019-04-06:
Looking forward to this feature.  Do you think scanning will be implemented soon?

dhalbert, 2019-04-06:
It will be after 4.0 is out. There are other BLE things to do, like BLE HID (which needs pairing/bonding), and we'll prioritize those after the 4.0 release.

rdagger, 2019-04-06:
Sounds great!  You guys are doing a fantastic job.

blademckain, 2019-05-07:
+1 need ble scanner! :)

dhalbert, 2019-07-31:
Implemented by #1993.

closed 2019-07-31 by dhalbert

rdagger, 2019-07-31:
That's great!  Thanks.

dhalbert, 2019-08-01:
This will be in 5.0.0-alpha.0, and is already available in the `master` builds here on S3: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin

rdagger, 2019-08-03:
The scanning works!  Here's my code:
```
import bleio
scanner = bleio.Scanner()
entries = scanner.scan(1)
```
I have a couple of requests.
1. Scan returns a lot of duplicate entries.  Would it be possible to only return unique entries?  Would this also speed up the scanning?
2. Address.address_bytes returns the address reversed.  Can it please be swapped?   On a side note, it looks like CircuitPython doesn't support reversing strings using [::-1].  It throws NotImplementedError only slices with step=1 (aka None) are supported.

Thanks again for all the great code.

dhalbert, 2019-08-04:
Check out the alpha release of https://github.com/adafruit/Adafruit_CircuitPython_BLE/releases/tag/2.0.0-alpha.0. `adafruit_ble.Scanner` has support for suppressing duplicates and other features. The `bleio` functionality is designed to be the minimum necessary, and we're layering Python code on top for end-user use in many cases.

The address bytes are as fetched from the advertisement. It will print in the correct direction, I think. I am trying to imitate the direction used by various BLE monitoring programs, but I will recheck.

rdagger, 2019-08-04:
The example code is very helpful, thanks!  Using the scanner library provides unique entries.
I'm pretty sure the byte order is reversed.  I created a simple test that flips the byte order:
```
from adafruit_ble.scanner import Scanner
from binascii import hexlify

address_types = {
    0: 'PUBLIC',
    1: 'RANDOM_STATIC',
    2: 'RANDOM_PRIVATE_RESOLVABLE',
    3: 'RANDOM_PRIVATE_NON_RESOLVABLE'}

scanner = Scanner()
entries = scanner.scan(3)  # scan for 3 seconds

for e in entries:
    address = hexlify(e.address.address_bytes).decode()
    # Reverse byte order and format address
    address = ':'.join(reversed([address[i] + address[i+1] for i in range(0, len(address), 2)]))
    print("{0}, RSSI: {1}, {2}".format(address,
                                       e.rssi,
                                       address_types[e.address.type]))
```

dhalbert, 2019-08-04:
if you just `print(e.address)`, does it print in the right order, for your purposes?  Doing `print()` or `str()` does reverse the bytes. The address_bytes are taken directly from the advertisement, in the order that they show up in the advertisement.

rdagger, 2019-08-04:
You're correct, `print()` and `str()` return the formatted address in the right order.
I was checking entries for addresses matching a known list and it wasn't obvious to me that I had to use `str()` with instances of the address class as opposed to address_bytes.  Thanks.