adafruit/circuitpython #1597: pulseio.PWMOut duty_cycle assignment fails infrequently

kevinjwalters, 2019-02-24:
I noticed this once during some manual testing and thought I'd imagined it. It then happened again and the value prints as 0 so it's not my imagination, perhaps occurs 1 in 30 assignments:

```
>>> a3pwmaout = pulseio.PWMOut(board.A3, duty_cycle=0, frequency=1000*1000, variable_frequency=True)
>>> ...lots of stuff in between...

>>> aout.value = 35*1000 ; a3pwmaout.duty_cycle = 0*1000
>>> aout.value = 0*1000 ; a3pwmaout.duty_cycle = 40*1000
>>> print(a3pwmaout.duty_cycle)
0
>>> aout.value = 0*1000 ; a3pwmaout.duty_cycle = 40*1000  ### 2nd attempt!
>>> print(a3pwmaout.duty_cycle)
39042
```

FYI, I'm using two other PWMOuts at same time which appear to be the maximum I can use :(

This is on `Adafruit CircuitPython 4.0.0-beta.2 on 2019-02-05; Adafruit CircuitPlayground Express with samd21g18`.

This is a pretty strange bug and it's infrequent so it could well be one that has existed for sometime. Is there any way for me to delve deeper when it occurs to gather more debug info?

dhalbert, 2019-02-24:
Is `aout` related to `a3pwmaout`? I notice it occurs when you set `aout.value = 0`.

kevinjwalters, 2019-02-24:
It's A0 in analogue out mode:

```
>>> import board
>>> from analogio import AnalogOut
>>> aout = AnalogOut(board.A0)
```

I just tried a loop iterating over some values

````
>>> for i in range(45):
...     wanted = i*1000
...     aout.value = 0*1000 ; a3pwmaout.duty_cycle = wanted ; print("wanted", wanted, "got", a3pwmaout.duty_cycle)
...     wanted = 0
...     aout.value = 35*1000 ; a3pwmaout.duty_cycle = wanted ; print("wanted", wanted, "got", a3pwmaout.duty_cycle)
...
...
...
wanted 0 got 0
wanted 0 got 0
wanted 1000 got 0
wanted 0 got 0
wanted 2000 got 1394
wanted 0 got 0
wanted 3000 got 2788
wanted 0 got 0
wanted 4000 got 2788
wanted 0 got 0
wanted 5000 got 4183
wanted 0 got 0
wanted 6000 got 0
wanted 0 got 0
wanted 7000 got 6971
wanted 0 got 0
wanted 8000 got 6971
wanted 0 got 0
wanted 9000 got 8366
wanted 0 got 0
wanted 10000 got 9760
wanted 0 got 0
wanted 11000 got 9760
wanted 0 got 0
wanted 12000 got 11154
wanted 0 got 0
wanted 13000 got 12549
wanted 0 got 0
wanted 14000 got 13943
wanted 0 got 0
wanted 15000 got 13943
wanted 0 got 0
wanted 16000 got 15337
wanted 0 got 0
wanted 17000 got 16732
wanted 0 got 0
wanted 18000 got 16732
wanted 0 got 0
wanted 19000 got 18126
wanted 0 got 0
wanted 20000 got 19521
wanted 0 got 0
wanted 21000 got 20915
wanted 0 got 20915
wanted 22000 got 20915
wanted 0 got 0
wanted 23000 got 22309
wanted 0 got 0
wanted 24000 got 23704
wanted 0 got 0
wanted 25000 got 23704
wanted 0 got 0
wanted 26000 got 25098
wanted 0 got 0
wanted 27000 got 26492
wanted 0 got 0
wanted 28000 got 27887
wanted 0 got 0
wanted 29000 got 27887
wanted 0 got 0
wanted 30000 got 29281
wanted 0 got 0
wanted 31000 got 30675
wanted 0 got 0
wanted 32000 got 30675
wanted 0 got 0
wanted 33000 got 32070
wanted 0 got 0
wanted 34000 got 33464
wanted 0 got 0
wanted 35000 got 34859
wanted 0 got 0
wanted 36000 got 34859
wanted 0 got 0
wanted 37000 got 36253
wanted 0 got 0
wanted 38000 got 37647
wanted 0 got 0
wanted 39000 got 37647
wanted 0 got 0
wanted 40000 got 39042
wanted 0 got 0
wanted 41000 got 40436
wanted 0 got 0
wanted 42000 got 41830
wanted 0 got 0
wanted 43000 got 41830
wanted 0 got 0
wanted 44000 got 43225
wanted 0 got 0
```

kevinjwalters, 2019-02-24:
Jumping by 5000 makes it easier to see failed non-0 assignments, e.g. `wanted 28000 got 0`:

```
>>> for s in range(0,5000,1000):
...     for w in range(s, 65000, 5000):
...         wanted = w
...         aout.value = 0*1000 ; a3pwmaout.duty_cycle = wanted ; print("wanted", wanted, "got", a3pwmaout.duty_cycle)
...         wanted = 0
...         aout.value = 35*1000 ; a3pwmaout.duty_cycle = wanted ; print("wanted", wanted, "got", a3pwmaout.duty_cycle)
...
...
...
wanted 0 got 0
wanted 0 got 0
wanted 5000 got 4183
wanted 0 got 0
wanted 10000 got 9760
wanted 0 got 0
wanted 15000 got 13943
wanted 0 got 0
wanted 20000 got 19521
wanted 0 got 0
wanted 25000 got 23704
wanted 0 got 0
wanted 30000 got 29281
wanted 0 got 0
wanted 35000 got 34859
wanted 0 got 0
wanted 40000 got 39042
wanted 0 got 0
wanted 45000 got 44619
wanted 0 got 0
wanted 50000 got 48802
wanted 0 got 0
wanted 55000 got 54380
wanted 0 got 0
wanted 60000 got 59957
wanted 0 got 0
wanted 1000 got 0
wanted 0 got 0
wanted 6000 got 5577
wanted 0 got 0
wanted 11000 got 9760
wanted 0 got 0
wanted 16000 got 15337
wanted 0 got 0
wanted 21000 got 20915
wanted 0 got 0
wanted 26000 got 25098
wanted 0 got 0
wanted 31000 got 30675
wanted 0 got 0
wanted 36000 got 34859
wanted 0 got 0
wanted 41000 got 40436
wanted 0 got 0
wanted 46000 got 44619
wanted 0 got 0
wanted 51000 got 50197
wanted 0 got 0
wanted 56000 got 55774
wanted 0 got 0
wanted 61000 got 59957
wanted 0 got 0
wanted 2000 got 1394
wanted 0 got 0
wanted 7000 got 6971
wanted 0 got 6971
wanted 12000 got 11154
wanted 0 got 0
wanted 17000 got 16732
wanted 0 got 0
wanted 22000 got 20915
wanted 0 got 0
wanted 27000 got 26492
wanted 0 got 0
wanted 32000 got 30675
wanted 0 got 0
wanted 37000 got 36253
wanted 0 got 0
wanted 42000 got 41830
wanted 0 got 0
wanted 47000 got 46013
wanted 0 got 0
wanted 52000 got 51591
wanted 0 got 0
wanted 57000 got 55774
wanted 0 got 0
wanted 62000 got 61351
wanted 0 got 0
wanted 3000 got 2788
wanted 0 got 0
wanted 8000 got 6971
wanted 0 got 0
wanted 13000 got 12549
wanted 0 got 0
wanted 18000 got 16732
wanted 0 got 0
wanted 23000 got 22309
wanted 0 got 0
wanted 28000 got 0
wanted 0 got 0
wanted 33000 got 32070
wanted 0 got 0
wanted 38000 got 37647
wanted 0 got 0
wanted 43000 got 41830
wanted 0 got 0
wanted 48000 got 47408
wanted 0 got 0
wanted 53000 got 52985
wanted 0 got 52985
wanted 58000 got 57168
wanted 0 got 0
wanted 63000 got 62746
wanted 0 got 0
wanted 4000 got 2788
wanted 0 got 0
wanted 9000 got 8366
wanted 0 got 0
wanted 14000 got 13943
wanted 0 got 0
wanted 19000 got 18126
wanted 0 got 0
wanted 24000 got 23704
wanted 0 got 0
wanted 29000 got 27887
wanted 0 got 0
wanted 34000 got 33464
wanted 0 got 0
wanted 39000 got 37647
wanted 0 got 0
wanted 44000 got 43225
wanted 0 got 0
wanted 49000 got 48802
wanted 0 got 0
wanted 54000 got 52985
wanted 0 got 0
wanted 59000 got 58563
wanted 0 got 0
wanted 64000 got 62746
wanted 0 got 0
```

kevinjwalters, 2019-02-24:
Ah, I intended to do that without the 0 assignments to just to make sure it still fails to assign when 0 isn't in play. 

```
>>> for s in range(0,5000,1000):
...     for w in range(s, 65000, 5000):
...         wanted = w
...         aout.value = 0*1000 ; a3pwmaout.duty_cycle = wanted ; print("wanted", wanted, "got", a3pwmaout.duty_cycle)
...
...
...
wanted 0 got 0
wanted 5000 got 4183
wanted 10000 got 9760
wanted 15000 got 13943
wanted 20000 got 19521
wanted 25000 got 23704
wanted 30000 got 29281
wanted 35000 got 34859
wanted 40000 got 39042
wanted 45000 got 44619
wanted 50000 got 48802
wanted 55000 got 54380
wanted 60000 got 59957
wanted 1000 got 59957
wanted 6000 got 5577
wanted 11000 got 9760
wanted 16000 got 15337
wanted 21000 got 15337
wanted 26000 got 25098
wanted 31000 got 30675
wanted 36000 got 34859
wanted 41000 got 40436
wanted 46000 got 44619
wanted 51000 got 50197
wanted 56000 got 55774
wanted 61000 got 59957
wanted 2000 got 1394
wanted 7000 got 6971
wanted 12000 got 11154
wanted 17000 got 16732
wanted 22000 got 20915
wanted 27000 got 26492
wanted 32000 got 30675
wanted 37000 got 36253
wanted 42000 got 41830
wanted 47000 got 46013
wanted 52000 got 51591
wanted 57000 got 55774
wanted 62000 got 61351
wanted 3000 got 2788
wanted 8000 got 6971
wanted 13000 got 12549
wanted 18000 got 16732
wanted 23000 got 22309
wanted 28000 got 27887
wanted 33000 got 32070
wanted 38000 got 37647
wanted 43000 got 41830
wanted 48000 got 47408
wanted 53000 got 52985
wanted 58000 got 57168
wanted 63000 got 62746
wanted 4000 got 2788
wanted 9000 got 8366
wanted 14000 got 8366
wanted 19000 got 18126
wanted 24000 got 23704
wanted 29000 got 27887
wanted 34000 got 33464
wanted 39000 got 37647
wanted 44000 got 43225
wanted 49000 got 48802
wanted 54000 got 52985
wanted 59000 got 58563
wanted 64000 got 62746
````

dhalbert, 2019-02-24:
Would you mind trying this with CircuitPython 3.1.2? That would help narrow things down? Thanks very much!

kevinjwalters, 2019-02-25:
Will try it with 3.1.2 in a day or two.

dhalbert, 2019-02-25:
OK, never mind about 3.1.2 - the code is the same. Looking at this.

dhalbert, 2019-02-25:
In your tests in the most recent comments, are you still using up a couple of other PWM's? Could you post the setup code for the tests? `PWMOut` may use a TC or a TCC timer, depending on what's available, and they are used differently.

One thing I will try is to try read the duty cycle a second time.

kevinjwalters, 2019-02-25:
This is the full code (from REPL):

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2 on 2019-02-05; Adafruit CircuitPlayground Express with samd21g18
>>>
>>> import board
>>> import digitalio
>>> import pulseio
>>> import analogio
>>> a0out = analogio.AnalogOut(board.A0)
>>> a1osc = pulseio.PWMOut(board.A1, duty_cycle=2**15, frequency=440, variable_frequency=True)
>>> a3pwmout = pulseio.PWMOut(board.A3, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=True)
>>> a6osc = pulseio.PWMOut(board.A6, duty_cycle=2**15, frequency=880, variable_frequency=True)
>>> a0out.value = 32768
>>> a0out.value = 0
>>> a3pwmout.duty_cycle = 0
```

And at the risk of losing a few I threw in an if to look for differences. It looks likely it's always the prior value. Three reads all show same value. I'm fairly certain the actual pwm wave doesn't change as that's how it was first apparent to me.

```
>>> maxdiff = 3100
>>> for s in list(range(0,5000,1000)) * 9:
...     for w in range(s, 65000, 5000):
...         a3pwmout.duty_cycle = w
...         read1 = a3pwmout.duty_cycle
...         read2 = a3pwmout.duty_cycle
...         read3 = a3pwmout.duty_cycle
...         if abs(w - read1) > maxdiff or abs(w - read2) > maxdiff or abs(w - read3) > maxdiff:
...             print("Write", w, "Read1", read1, "Read2", read2, "Read3", read3)
...
...
...
Write 50000 Read1 42740 Read2 42740 Read3 42740
Write 7000 Read1 0 Read2 0 Read3 0
Write 12000 Read1 0 Read2 0 Read3 0
Write 54000 Read1 48438 Read2 48438 Read3 48438
Write 35000 Read1 28493 Read2 28493 Read3 28493
Write 58000 Read1 51288 Read2 51288 Read3 51288
Write 14000 Read1 8548 Read2 8548 Read3 8548
Write 36000 Read1 28493 Read2 28493 Read3 28493
Write 18000 Read1 11397 Read2 11397 Read3 11397
Write 31000 Read1 25644 Read2 25644 Read3 25644
Write 38000 Read1 31342 Read2 31342 Read3 31342
Write 54000 Read1 48438 Read2 48438 Read3 48438
Write 51000 Read1 45589 Read2 45589 Read3 45589
Write 43000 Read1 37041 Read2 37041 Read3 37041
Write 64000 Read1 56986 Read2 56986 Read3 56986
Write 24000 Read1 17096 Read2 17096 Read3 17096
Write 22000 Read1 14246 Read2 14246 Read3 14246
Write 42000 Read1 34192 Read2 34192 Read3 34192
Write 39000 Read1 31342 Read2 31342 Read3 31342
Write 61000 Read1 54137 Read2 54137 Read3 54137
```

Semi-related, if you have any magic for getting one more PWM output in CircuitPython, please share! See: https://forums.adafruit.com/viewtopic.php?f=60&t=148017

dhalbert, 2019-02-28:
I have reproduced this on a Metro M0 with this test program:
```
import board, pulseio

d9_pwm = pulseio.PWMOut(board.D9, duty_cycle = 2**15, frequency = 2*1000*1000, variable_frequency=True)

while True:
    prev_read_dc = -1
    prev_dc = -1
    for dc in range(0, 60000, 10000):
        d9_pwm.duty_cycle = dc
        read_dc = d9_pwm.duty_cycle
        if read_dc == prev_read_dc:
            print('requested: {}, prev requested: {}, read: {}, prev read: {}'.format(
                dc, prev_dc, read_dc, prev_read_dc))
        prev_read_dc = read_dc
        prev_dc = dc
```
It only prints anomalies. Reducing the frequency reduces the errors greatly. The pin is using a TCC. Will now try to debug.

dhalbert, 2019-03-01:
@kevinjwalters Could you please test this uf2? It seems to fix the above problem. But I have not actually looked at the output waveforms, and you're in a better position to do that. I tested on a Metro M0, but I believe I replicated the test conditions you set up (I used the same physical chip pin, etc.). Thanks.

[cpx-fix-pwmout-tcc-2019-02-28.uf2.zip](https://github.com/adafruit/circuitpython/files/2917474/cpx-fix-pwmout-tcc-2019-02-28.uf2.zip)

kevinjwalters, 2019-03-01:
Sure, I can give it a test today or tomorrow on my CPX. The explanation in #1605 fits the type of bug I was expecting particularly wrt sporadic occurrence.

Does that uf2 need to re rebuillt due to the absent bit that @sommersoft noted in #1605?

dhalbert, 2019-03-01:
> Does that uf2 need to re rebuillt due to the absent bit that @sommersoft noted in #1605?
No, that just broke the M4 builds (syntax error). The CPX build I gave you above is fine.

kevinjwalters, 2019-03-01:
Ran my previous test, then ran it again with upping the iterations from 9 to 99 - all tiptop - fix looks good on my CPX, ta.

dhalbert, 2019-03-01:
Great, are you getting the waveforms out that you want?

dhalbert, 2019-03-01:
Fixed by #1605

closed 2019-03-01 by dhalbert

kevinjwalters, 2019-03-02:
Waveforms are good, just been playing with those.