adafruit/circuitpython #1598: DIPSIO: add way to check if dispio object is being shown

TG-Techie, 2019-02-24:
or an _onshow_ method to for touchable objects, etc

TG-Techie, 2019-02-24:
that had unscored on either side

siddacious, 2019-02-24:
@TG-Techie Please clean up your initial request/comment to include more information about what you're trying to achieve and how what you're asking for will help achieve your goals.

TG-Techie, 2019-02-24:
my apologies, more detail for not issues too. good explanation?

siddacious, 2019-02-24:
Much better, thank you.

TG-Techie, 2019-02-24:
how do you add labels? or is it a permissions thing?

tannewt, 2019-03-08:
We should add a function to Group that returns the child with the opaque pixel from a given coordinate.

TG-Techie, 2019-03-08:
Agreed, "fetch", "pull", "of"?

dhalbert, 2019-03-08:
`child_at` ?

TG-Techie, 2019-03-08:
When you say child do you mean a pixel color or an object?

kmatch98, 2020-09-06:
@tannewt I’m interested in working on this one. 

There should be a way of tapping into the display redrawing code and the display elements `get_pixel` routines you determine which is opaque and take advantage of any x,y and mirror, rotation transform code that is already there. If you suggest an approach I will try to figure out how to make it work. 

As for operation of this `child_at(x,y)` function, how should it deal with a child element that is a Group?

1. If the child Group is at that pixel should it return the child Group and you would have to keep iterating to get to the actual display object?  If so, the `child_at` function will need to accept a Group as an optional argument so that you could iterate on a child Group. 

2. Or else should `child_at` iterate all the way down through any Group hierarchy and find the display element that is a “leaf” of the hierarchy tree?

tannewt, 2020-09-08:
@kmatch98 That's a great question! My first instinct is to return a TileGrid that matched. However, the group may actually be the best unit. Perhaps the missing piece is adding a `depth` kwarg that could be set to indicate that a group is fine. I imagine that would be best for a button grid for example.

tannewt, 2020-09-10:
I don't think it's worth doing in Python. Instead, `layer_at` could be a function on the native Group object. It'll be faster and have raw access to displayio internal APIs.

I do think you'll want to handle transparency.

To handle fuzzy touch you could call layer_at for surrounding pixels as well and then decide amongst all of the results what was touched.