adafruit/circuitpython #1599: displayio documentation typos

dhalbert, 2019-02-25:
1. https://github.com/adafruit/circuitpython/blob/master/shared-bindings/displayio/ParallelBus.c#L42 says
```
//| :class:`ParallelBus` -- Manage updating a display over SPI four wire protocol
//| ==============================================================================
//|
//| Manage updating a display over SPI four wire protocol in the background while Python code runs.
```
That's kind of the same text in `FourWire.c`, but is that a copy/paste typo?

2. https://github.com/adafruit/circuitpython/blob/master/shared-bindings/displayio/__init__.c#L74 says:
```
//| All libraries change hardware state but are never deinit
```
It's cut off, should be `deinited.`, and maybe more?

dhalbert, 2019-02-25:
3. `FourWire` should note that the `reset` pin arg is defaulted to `None`:
```
 FourWire(spi_bus, *, command, chip_select, reset=None)
```
The text below the defn could also note that.

tannewt, 2019-03-08:
Please also add a milestone. I may miss it if it doesn't have one even if it is assigned to me.

commit afbf5901 (2019-03-13): Update displayio docs to add detail to display bus comments

Fixes #1599
closed 2019-03-13 by dhalbert