adafruit/circuitpython #1609: USB disconnect or reset mutes sound on the computer

deshipu, 2019-03-01:
For some reason with the 4.0 code, even with ``CIRCUITPYTHON_USB_HID=0`` and ``CIRCUITPYTHON_USB_MIDI=0``, disconnecting or hard-resetting the board results in my sound getting muted. I tested that initially with PewPew (which is pretty much a Trinket M0), but confirmed it now with a Hallowing M0 Express.

Interestingly enough, nothing is appearing in ``xev`` when that happens.

dhalbert, 2019-03-01:
This is on Ubuntu, right? We should test on other OS's.

siddacious, 2019-03-01:
I've been plugging/unplugging, resetting etc. ad nauseam the last few days with master+ and haven't noticed anything getting muted on my Mac (Mojave)

deshipu, 2019-03-01:
Yes, more precisely Linux Mint with XFCE4. It doesn't happen on Fedora 28, also with XFCE4.

deshipu, 2019-03-01:
I think that it is not keyboard — I disabled keyboard shortcuts for volume control, and it still happens.

dhalbert, 2019-03-01:
are they regular keyboard shortcuts or Consumer Control shortcuts?

deshipu, 2019-03-01:
How do I tell?

I just checked that it doesn't happen with 4.0.0-alpha.5

deshipu, 2019-03-01:
Looks like commit b5e40f52c2b1c2890c866d7305e0e99b4811b7f5 is the problem.

dhalbert, 2019-03-01:
Thanks for the bisect or equivalent. That's a big commit: USB MIDI and a lot of tinyusb changes. I'll put a sniffer on the USB traffic and see what's up.

deshipu, 2019-03-01:
Another thing that may be relevant: the board shows up in alsamixer in the list of sound cards. My current working theory is that Ubuntu does something weird when a sound card is disconnected — perhaps the same mechanism that switches volume when headphones are connected/disconnected.

deshipu, 2019-03-01:
I wonder how hard it would be for the board not to appear as a MIDI device when ``CIRCUITPYTHON_USB_MIDI`` is disabled, and also not appear as a keyboard when ``CIRCUITPYTHON_USB_HID`` is disabled.

dhalbert, 2019-03-01:
Then it's the MIDI. USB MIDI is a subclass of audio device (from https://github.com/adafruit/circuitpython/blob/master/tools/gen_usb_descriptor.py)
```
audio_midi_interface = standard.InterfaceDescriptor(
    description="Midi goodness",
    bInterfaceClass=audio.AUDIO_CLASS_DEVICE,
    bInterfaceSubClass=audio.AUDIO_SUBCLASS_MIDI_STREAMING,
    bInterfaceProtocol=audio.AUDIO_PROTOCOL_V1,
    iInterface=StringIndex.index("CircuitPython MIDI"),
```

>I wonder how hard it would be for the board not to appear as a MIDI device when `CIRCUITPYTHON_USB_MIDI` is disabled, and also not appear as a keyboard when `CIRCUITPYTHON_USB_HID` is disabled.

We have an open issue (#1015) for dynamic USB descriptors. We could possibly conditionalize the static descriptor in the file above for now.

kevinjwalters, 2019-03-01:
I found the CPX didn't work as a keyboard when I plugged it into a school's computer. It made me wonder if there was something about the other usb devices that it was showing up as which made it block the port. It's alluded to in https://forums.adafruit.com/viewtopic.php?f=48&t=146648#p725650

I'd tweaked the CPX dice project (https://github.com/kevinjwalters/circuitpython-examples/blob/master/cpx/cpx-dice-typer.py) and it was supposed to type the numbers. Worked at home but not at the school I was volunteering at. I didn't have time to delve deeper. I might be able to revisit this but I suspect they'll be a lack of expertise _on site_.

deshipu, 2020-09-02:
I think this is now fixed on the system side, and we can close it.

closed 2020-09-02 by deshipu

kevinjwalters, 2020-09-05:
@deshipu What do you mean, was this a bug on the linux side of things?

deshipu, 2020-09-05:
Yes, Ubuntu and derivatives would reset the sound level to zero when a new MIDI device was connected. CircuitPython devices happen to be MIDI devices.