adafruit/circuitpython #1612: displayio and SPI conflict

jerryneedell, 2019-03-02:
There appears to be some conflict with displayio and SPI.
I have a TFT featherewing on a Feather_m4_express ( same issue on a particle xenon)
and if I run the stmpe610_simpletest which reads the TFT touch screen it works normally as long as I have not configured displayio first. If set up diaplayio -- so the REPL is being echoed to the screen, and then try the stmpe610_simpletest, it crashes.

PR #1604 had no impact on this -- it behaved the same with or without it.

normal run
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-142-gced37c100 on 2019-03-02; Adafruit Feather M4 Express with samd51j19
>>> import stmpe610_simpletest
Go Ahead - Touch the Screen - Make My Day!
(1095, 2189, 62)
(1173, 2340, 21)
(1141, 2350, 25)

```
now configure displayio and reboot with REPL on display
and repeat
```
>>> import display

soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-142-gced37c100 on 2019-03-02; Adafruit Feather M4 Express with samd51j19
>>> import stmpe610_simpletest
Go Ahead - Touch the Screen - Make My Day!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "stmpe610_simpletest.py", line 9, in <module>
  File "adafruit_stmpe610.py", line 230, in buffer_empty
  File "adafruit_stmpe610.py", line 178, in _read_byte
  File "adafruit_stmpe610.py", line 308, in _read_register
  File "adafruit_stmpe610.py", line 304, in _read_register
RuntimeError: Function requires lock
>>> 

```

jerryneedell, 2019-03-03:
a followup - the issue does not occur if I execute 
```displayio.release_displays()```
then run the stmpe610_simpletest

dhalbert, 2019-03-03:
I believe tjhe displayio object (and its SPI object) and the pin assignments live past a soft reload, so that the screen can show the text of the reload. I'm not sure if your issue is associated with this, but it's something to keep in mind.

dhalbert, 2019-03-05:
The TFT FeatherWing uses the regular SPI pins to talk to the display. `"stmpe610_simpletest.py` does:
```
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
```
So since those pins are already in use by `displayio`, you'll get an error.

jerryneedell, 2019-03-05:
I changed it to use board.SPI()  so I don't get the "pin in use" error.

dhalbert, 2019-03-05:
`st.read_data()` eventually uses BusDevice, which should wait for and then grab the lock for the SPI bus, locking out displayio while reading. Somehow someone is violating the lock protocol. We'll have to read the displayio code to look at this.

dhalbert, 2019-03-05:
OK, I believe this problem is due to #1619.

jerryneedell, 2019-03-06:
some progress with current master build and SPI 
current master build allow me to mount/access the SD CArd with displayio active 
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Adafruit CircuitPython 4.0.0-beta.2-195-g233cfcd4a on 2019-03-05; Adafruit Feather nRF52840 Express with nRF52840
>>> 
>>> 
>>> import sdmount_lib
>>> import hello
Howdy
>>> 
```  in the above case hello.py is on the SD Card
sdmount_lib.py
```
import adafruit_sdcard
import busio
import digitalio
import board
import storage
import sys
# Connect to the card and mount the filesystem.
cs = digitalio.DigitalInOut(board.D5)
sdcard = adafruit_sdcard.SDCard(board.SPI(), cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
sys.path.append("/sd")
sys.path.append("/sd/lib")
```

PR #1620 (2019-03-07): Rework displayio display bus transaction handling during refresh
Success or failure of starting a display-bus transaction is now checked. For some routines, we busy-wait for the bus. For the background display refresh task, we just give up if the bus is not currently available.

Meant to fix #1612 and #1619.
closed 2019-03-07 by dhalbert