adafruit/circuitpython #1613: TileGrid position has some unknown transformation (and also we can't move)

ladyada, 2019-03-02:
this could be me not understanding tilegrid but when i subclass it to create shapes with outlines, it displays in the right spot, but the position is weirdly transformed?

download this:
https://github.com/ladyada/Adafruit_CircuitPython_Display_Shapes/blob/master/adafruit_display_shapes/rect.py
then run:

```import displayio
import board
from adafruit_display_shapes.rect import Rect

# Make the display context
splash = displayio.Group()
board.DISPLAY.show(splash)
rect = Rect(10, 10, 20, 20, fill=0xFF0000, outline=0x0000FF)
splash.append(rect)

print(rect.position)

while True:
    pass
```

red rect with blue outline appears, but position prints out as `(13504, 8192)` not `(10,10)`

Running `Adafruit CircuitPython 1ba7610 on 2019-03-01;`

zencuke, 2019-03-12:
I don't know if this adds any information but as long as I did the test I thought I would add it here. I was looking at this code as a tilegrid example (I'm hoping to create similar app specific drawing primitives) and got curious. I added a bunch of debug print statements to rect.py:rect.__INIT__ to see where it happened. The position value was initially (0,0) and changed after the line 

> self._bitmap =displayio.Bitmap(...) 

to the same value printed in the above wrapper example code.

zencuke, 2019-03-12:
The exact debug statement I added to the various places in rect.py was.

> print (self.position)

tannewt, 2019-03-12:
I'll be looking at this today because I'm going to switch position to separate x and y properties to make TileGrid consistent with Group.

commit 224e9b10 (2019-03-13): Standardize TileGrid to x and y properties over position

This brings it inline with Group. Also fixes #1613

This also includes a number of fixes for where a method is called
through a subclass. We now correctly get the native object.
Fixes #1567

Lastly, this adds subscript support to TileGrid for changing tile
indices. Similar to Bitmap, it accepts ints or 2-tuples.
closed 2019-03-13 by dhalbert