adafruit/circuitpython #1615: particle xenon safe mode with tft featherwing

jerryneedell, 2019-03-03:
I have been trying to use a particle xenon with a tft featherwing to test displayio.
In general, it works fine except that very often I on power up, the system goes into SAFE mode due to a low power condition.  I have not seen this with the feather  nrf52840 or feather_m4_express.

It usually recovers and boots   normally if I just press RESET so it appears to be related to the power up.

Since the issue does not occur with the feather nrf52840 it appears it is due to the power supply on the particle board,  so feel free to close this if it is not appropriate.

I thought it might be of interest and I was curious if anyone has run into similar issues.

dhalbert, 2019-03-03:
It might be that the power supply design is such that there's more of delay than on the feather before the the voltage gets above 2.7V, which is the brownout threshold. We could wait some number of milliseconds before enabling the brownout detection. This is in `port_init() in `ports/nrf/supervisor/port.c`.

jerryneedell, 2019-03-03:
Thanks! I will experiment with it.

jerryneedell, 2019-03-03:
hmm -- I tried adding delays from 2 to 10 ms and it made no difference but I found something else of interest.  Grounding the EN pin on the particle_xenon only drops the 3.3V signal to 1V ?? It goes to 0V on a Feather_m4 (and I assume on all feathers)  Both rise to 3.3 V in a few hundred microseconds but does the threshold test possibly get confused because the xenon is starting at 1V?  In any case - having EN not working properly is troublesome. That is how the TFT featherwing controls power to the board.

I verified the same behavior even without the TFT featherwing installed.

jerryneedell, 2019-03-04:
a few more observations:
I checked a particle argon and also found that the 3.3V only goes to 1V when EN is grounded.
Interestingly, when I did that, the argon rebooted to Safe Mode!

The Safe mode boot only appears to occur if the EN pin is used to power cycle the xenon board on the TFT featherwing. It boots reliably if I just apply power via the USB port.

I will post a question to the Particle.IO forum.

jerryneedell, 2019-03-06:
AS noted by @sommersoft it looks like the power regulator on the Xenon is working as desighend
According to the datasheet the output voltage only goes to ~.8 V when the CE (EN pin)  is Low.

[image]

So, the question remains, why does CP go to Safe Mode when the EN Pin is brought back to High? Is this something that should be "fixed" in CP or is it because the Particle boards are not operating in an acceptable way.

Is  it problematic to use have a Xenon connected to a featherwing board with the 3.3V sitting at .8V for a long period of time. Should we avoid using the EN pin as an On/off for the Particle Boards?

tannewt, 2019-08-23:
This still happen?

jerryneedell, 2019-08-23:
Yes it does. Not surprising since it appears to be a design issue on the Xenon.

tannewt, 2019-08-23:
Is there anything we can do about it? If not, can we close this?

jerryneedell, 2019-08-23:
Its something to be aware of if anyone runs into it, but it's not clear there is anything to be done in CP to "fix it".  I'm happy with closing it.

closed 2019-08-23 by jerryneedell