adafruit/circuitpython #1616: main.py CircuitPython 3.1.2 on 2019-01-07 runs out of memory

silentShadow, 2019-03-04:
Opened up new M0 express and updated to CircuitPython 3.1.2 from the latest Repo and get the following:

Traceback (most recent call last):
  File "main.py", line 14, in <module>
  File "adafruit_hid/keyboard.py", line 35, in <module>
  File "adafruit_hid/keycode.py", line 31, in <module>
  File "adafruit_hid/keycode.py", line 195, in Keycode
MemoryError: memory allocation failed, allocating 776 bytes

Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Metro M0 Express with samd21g18

dhalbert, 2019-03-04:
Which version of the `adafruit_hid` library did you load? You need to use the compiled `.mpy` 3.x version from here: https://github.com/adafruit/Adafruit_CircuitPython_HID/releases/latest. Otherwise you'll run out of memory when importing. And if you have both the .py and .mpy versions, the .py will take precedence.

silentShadow, 2019-03-04:
I'm honestly not sure, it's just what exists in the main.py file on my device.  I didn't modify it.  I did a flash erase and removed the main.py file for now.  I'll grab the version you linked and see what happens.

dhalbert, 2019-03-14:
@silentShadow Are you still having trouble?

dhalbert, 2019-03-18:
Closing for now. Good places for this support are in https://adafru.it/discord, #help-with-circuitpython channel, or https://forums.adafruit.com.

closed 2019-03-18 by dhalbert