adafruit/circuitpython #1623: Allow looping audio samples to be stopped after a complete sample

kevinjwalters, 2019-03-08:
It would be useful to allow audio samples that are playing with the looping feature to be _optionally_ terminated after a complete sample, e.g. to an integer number of samples will be repeated. I've not considered the case for non-looping but perhaps that would also be a desirable optional feature for some use cases.

A carefully constructed looping sample this would start at "0" (midpoint would be 32768 for 16bit dac) and end at or very near there. The current system stops immediately and will leave the DAC at an unknown sample probably away from the the midpoint, see discussion in comments of #1090. If the users wishes to play short samples back to back or waveform based samples this would be useful to avoid minor delays (due to slew rate) or pops (probably won't happen due to the ramping introduced by #1090).

If there are issues for this feature specific to M0 or M4 boards it would be useful to mention them in documentation.

kevinjwalters, 2020-06-09:
Is there a hacky way of doing this, e.g. 

```
while True:
    if dac.atend() or dac.nearend():  ### can this be done somehow with current implementation?
        dac.stop()
        break
```

tannewt, 2020-06-09:
Not that I know of.