adafruit/circuitpython #1625: Add support for splitting screen with displayio

jpecor, 2019-03-09:
It would be nice to allow partitioning/splitting of the display with displayio such that you could use each partition as an independent display.  For example, drawing graphics - buttons, shapes, etc. - on the top half while preserving the REPL output on the bottom of a single display.

I'm admittedly fuzzy on how the layering works with displayio Groups right now, so maybe we're not far from this capability, already.

ladyada, 2019-03-09:
yeah this could def be handy - probably easiest to do by being able to define the size/location of the splash

tannewt, 2019-03-09:
I'm thinking it'll be more like a second terminal output that you can connect to the "stdout" of CircuitPython. I want to make the input and output streams more generic in the future. (For example, keyboard support is the reverse of this.)

siddacious, 2019-03-09:
@tannewt  I like that idea

tannewt, 2019-03-12:
FYI I came across this testing code of mine:

```python
p = displayio.Palette(2)
p.make_transparent(0)
p[1] = 0xff0000
w, h = terminalio.FONT.get_bounding_box()
tilegrid = displayio.TileGrid(terminalio.FONT.bitmap, pixel_shader=p, x=10, y=0, width=30, height=3, tile_width=w, tile_height=h)
print(tilegrid.x, tilegrid.y)
t = terminalio.Terminal(tilegrid, terminalio.FONT)
t.write("terminal test\r\nover\r\nother text")
g.pop(1)
g.append(tilegrid)
```

While it doesn't display CircuitPython's output, it would allow you to display your own text in a terminal. Is that enough for you @jpecor ?

jpecor, 2019-03-12:
Thanks @tannewt!    A couple of questions:  

I replaced "x=10, y=0" with "position=(10,0)".  I presume that was the intent of the original code?

Also, what is "g" supposed to be?  There's no original declaration/constructor for it, so I get an error.

tannewt, 2019-03-12:
ah, ya. I have a PR out to change position to separate x and y. So position will work now and x and y in the future.

g is the group you want the terminal to be in. Terminal really just modifies the TileGrid for you.

This is my full file but it has extra stuff:

```python
import adafruit_ili9341
from adafruit_display_text import text_area
from adafruit_bitmap_font import bitmap_font
import terminalio
import board
import displayio
import struct
import time
import supervisor

# This example runs on a Feather M4 Express or the Feather nRF52840 Express with the 2.4" TFT
# FeatherWing: https://www.adafruit.com/product/3315
# It also requires the Adafruit_CircuitPython_ILI9341 library.

# We must release any active displays before we use them.
displayio.release_displays()

# The display bus carries commands to the display.
display_bus = displayio.ParallelBus(data0=board.A3, command=board.A9, chip_select=board.A10, write=board.A8, read=board.A11)
# display_bus = displayio.FourWire(board.SPI(), command=board.D10, chip_select=board.D9)

# This manages drawing the active group to the display. It also manages the backlight
display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240, rotation=270)
display_bus.send(0x37, struct.pack(">H", 0))

print("hello world")
print(terminalio.FONT.bitmap.width, terminalio.FONT.bitmap.height)

g = displayio.Group(max_size=10, scale=2)
print("loading static")
text = text_area.TextArea(terminalio.FONT, width=20, height=20, x=10, y=10)
text.text = "Static"
g.append(text)

display.show(g)

time.sleep(1)
text.text = "Static but longer"

time.sleep(1)
text.x += 10
text.y += 10
text.text = "Static short"

print("loading dynamic")
font = bitmap_font.load_font("/scientifica-11.bdf")

dynamic_text = text_area.TextArea(font, width=20, height=20, x=100, y=100)
dynamic_text.text = "Dynamic"
g.append(dynamic_text)

p = displayio.Palette(2)
p.make_transparent(0)
p[1] = 0xff0000
w, h = terminalio.FONT.get_bounding_box()
tilegrid = displayio.TileGrid(terminalio.FONT.bitmap, pixel_shader=p, x=10, y=0, width=30, height=3, tile_width=w, tile_height=h)
print(tilegrid.x, tilegrid.y)
t = terminalio.Terminal(tilegrid, terminalio.FONT)
t.write("terminal test\r\nover\r\nother text")
g.pop(1)
g.append(tilegrid)

g.scale = 3

time.sleep(2)

print(tilegrid[0], tilegrid[0,0], tilegrid[1,1])
tilegrid[0] = 10
time.sleep(1)
tilegrid[0,0] = 11
time.sleep(1)
tilegrid[1,1] = 20

time.sleep(60)
```

jpecor, 2019-03-13:
Ah!  I was missing this line:

`g = displayio.Group(max_size=10, scale=2)`

The position=(x,y) feels intuitive to me.

tannewt, 2019-03-13:
Ya, I was torn about it. It's slightly more efficient to use numbers instead of tuples so that's been my bias. I wonder if I'm compromising my intuitive over efficient goal though...

jpecor, 2019-03-13:
If the x=, y= approach is more optimized, I'd favor that route, as well.  Less Pythonic, maybe, but for CircuitPython every bit saved will likely matter.     Also, I must have some other libs that are long in the tooth.  My version of Text doesn't have x, y in __init__.

Just checked GitHub, and TextArea is gone.  Replaced with Label?   Wow, things move fast!

tannewt, 2019-03-13:
Yup! We're moving fast to get things as we want for 4.0.0 stable. Thanks for joining us on the ride!