adafruit/circuitpython #1626: pulseio.PWMOut() instantiation fails sometimes with constant same frequencies on CPX

kevinjwalters, 2019-03-09:
I'm seeing some rather non-deterministic behavior with creation of `pulseio.PWMOut()`. I've been discussing how to get the most PWM outputs on a Circuit Playground Express: https://forums.adafruit.com/viewtopic.php?f=60&t=148017. I can sometimes get four by using the same frequency and `variable_frequency=False` but not always! Example below is a variant of 4.0.0 beta 2 with no `code.py`/`main.py` and after a power off/on to rule out any accumulated-weirdness(tm):

```
Adafruit CircuitPython 4.0.0-beta.2-141-g2c9fbb5d4-dirty on 2019-02-28; Adafruit                                CircuitPlayground Express with samd21g18
>>> import board, pulseio
>>> p3 = pulseio.PWMOut(board.A3, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
[... 87 lines cut ...]
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.2-141-g2c9fbb5d4-dirty on 2019-02-28; Adafruit CircuitPlayground Express with samd21g18
>>> import board, pulseio
>>> p2 = pulseio.PWMOut(board.A2, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
>>> p3 = pulseio.PWMOut(board.A3, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: All timers in use
>>> p3 = pulseio.PWMOut(board.A3, duty_cycle=2**15, frequency=2*1000*1000, variable_frequency=False)
```

Note in that last attempt where a "retry" surprisingly works, the order is reversed for p2 & p3 instantiation. Although in general, I've had more success with p3 first then p2.

I'd welcome a workaround too if this can't be fixed in 4.0.0. Perhaps I could code it up so it makes p2, then tries p3 ten times and if that doesn't work then then cleans up, makes p3 then tries p2 ten times?

kevinjwalters, 2019-03-12:
For what it's worth I tried it on `4.0.0-beta.3` and same fault occurs there.

I've written some python code that now tries to initialise them with retries and then reverses order. My current implementation doesn't seem to be that useful as a workaround. Control-C/Control-D for a soft reset will eventually work but it often takes ten attempts :(

dhalbert, 2019-03-12:
Try this uf2. I made one minor change to wait for a reset of a TCC before proceeding. But I don't see anything else wrong, like uninitialized variables.
[cpx-tcc-wait-2019-03-11.uf2.zip](https://github.com/adafruit/circuitpython/files/2954845/cpx-tcc-wait-2019-03-11.uf2.zip)

dhalbert, 2019-03-12:
Also, if that doesn't fix things, I'd be interested to know if you see any difference when using a lower frequency (like 2000 instead of 2000000)

kevinjwalters, 2019-03-12:
`4.0.0-beta.3-4-g186e31591-dirty` doesn't appear to make any difference, I still get initialisation errors. I'll have a play with frequency later. For lower frequency should l I use `4.0.0 beta 3` or `4.0.0-beta.3-4-g186e31591-dirty`?

dhalbert, 2019-03-12:
The uf2 I gave you fixes something that's probably inconsequential, but I would keep that change anyway, so test against that.

Have you looked at the `pins.c` file for CPX to get the actual PA/PBxx pin numbers? Then you can look up which TC's or TCC's are available for each PAxx or PBxx.

kevinjwalters, 2019-03-12:
From some brief testing, 2kHz seems like similar behaviour with many failures using `4.0.0-beta.3-4-g186e31591-dirty`.

I can't see a clear pattern yet, maybe a power off makes it more likely to initialise ok but it's not on the first execution of code.py as one might expect. The second attempt (i.e. power off/on, auto-execute code.py, soft reset post exception, second code.py execution) seems pretty successful though.

dhalbert, 2019-03-13:
OK, it was an arg processing bug. Even though you said `variable_frequency=False`, it was taking it as `True`, which made it use only TCC's. Try this: 
[cpx-pwmout-arg-fix-2019-04-12.uf2.zip](https://github.com/adafruit/circuitpython/files/2959365/cpx-pwmout-arg-fix-2019-04-12.uf2.zip)

kevinjwalters, 2019-03-13:
I've just tested that latest version, `4.0.0-beta.3-25-g2169a6240-dirty`, and it worked first time and has worked for nine soft reboot (Control-C / Control-D). Previous versions had roughly a 9/10 failure rate. Looks good.

kevinjwalters, 2019-03-13:
I just using `4.0.0-beta.3-25-g2169a6240-dirty` for a few minutes and then left it. When I came back and tried to connect to the serial (over USB) it didn't work. The drive (on Windows) was no longer showing up with the typical spaced used but just marked as `FAT` and opening it in explorer showed it as empty. It responded to reset button but that's left me with an `CIRCUITPY` drive that's _missing my code_! I've only got some wavs, a `README`, `.fseventsd` and `.Trashes`. I did a power cycle and file system still looks the same, no sign of my code.py etc.

I think we can rule out user error here because the whole lib directory has gone AWOL and that takes ages to delete so I would have noticed if I'd done this accidentally.

Is there anyway the file system can be _auto_ reinitialised?

kevinjwalters, 2019-03-13:
It's getting weirder, the wavs have disappeared and a `.metadata_never_index` has appeared. Minor chance this is related to Patch Tuesday but that's even more frightening.

kevinjwalters, 2019-03-13:
My boot_out currently have in it 

```
Adafruit CircuitPython 4.0.0-beta.3-25-g2169a6240-dirty on 2019-03-12; Adafruit CircuitPlayground Express with samd21g18```

but earlier it had a large hex number with some hypens with braces on either end.

kevinjwalters, 2019-03-13:
I've decided to go back to 4.0.0 beta 3 to see if that's stable. I put that on and this popped out when I connected to serial console afterwards:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.0.0-beta.3 on 2019-03-07; Adafruit CircuitPlayground Express with samd21g18
>>>
```

`boot_out` currently contains:

```
Adafruit CircuitPython 4.0.0-beta.3-25-g2169a6240-dirty on 2019-03-12; Adafruit CircuitPlayground Express with samd21g18
```

and has a date of 1-Jan-2000 which is a bit odd. I'm going to `storage.erase_filesystem()` to ensure I've got a clean slate.

closed 2019-03-13 by tannewt

kevinjwalters, 2019-03-17:
4.0.0 beta 5 looks good, btw.

PR #1641 (2019-03-13): wrong arg type for PWMOut variable_frequency
PWMOut was reading a supplied bool arg as an int, so it was always True.

Also minor fix to make sure SWRESET is finished before proceeding.

Fixes #1626.