adafruit/circuitpython #1627: [nrf52840] Filesystem errors when virtual drive fills up

uhrheber, 2019-03-09:
Tested boards: pca10056, pca10059
Steps to reproduce the errors:
- take a blank board, connect it to a computers USB, put a simple main.py on it (LED blink, or such) and run it
- create a subdirectory on the virtual drive (e.g. /lib)
- copy some CircuitPython libs into the subdir (a few 10kB of files are usually enough, but sometimes more is needed)

The following errors occur, sometimes immediately, sometimes after disconnecting and reconnecting the board from the USB:
- script crashes and shows reading errors in main.py
- all files on the virtual drive become read only
- some files on the virtual drive become corrupted or deleted
- the whole drive is wiped and appears freshly formatted

This happens with all  4.x beta releases, as well with self-compiled versions. Increasing the stack size doesn't help.

Could anybody please try to reproduce this? Maybe also on other boards?

uhrheber, 2019-03-14:
BTW: How circuitpython is flashed doesn't make a difference.
I flashed boards directly with J-Link (without a bootloader), with the Adafruit UF2 bootloader, and with the Nordic DFU bootloader.

kevinjwalters, 2019-03-14:
What o/s including version is running on the host computer? Why does title mention "drive fills up"?

uhrheber, 2019-03-15:
I tested several host computers, with Windows 10 and Linux Mint 18.3/19.1. Same results.

"Drive fills up" means, the fuller the virtual drive gets (regardless what data, may be Python code, random test data), the more likely the read errors appear.

As I said, a single Circuitpython file may run without an error indefinitely, but as soon as you copy additional files to the virtual drive, the errors appear. If the drive is nearly full, the errors occur inevitably.

kevinjwalters, 2019-03-15:
That's interesting detail. I was only asking because a disk/storage device being completely full often causes various forms of mayhem and I wasn't sure if you were talking about 100% full or now.

tannewt, 2019-03-27:
I'm unable to reproduce this with 94822ee2088960e6f68c9d4f05606cdb47c9ba5c built in debug mode on the PCA10056. I loaded it with JLink and then added a blinky code.py (below), made the lib folder and then dragged the lib folder over. I also copied the examples over. It was slow but didn't have errors.

Next I repeatedly ejected the drive, disconnected the usb data lines with an inline switch, reset circuitpython via GDB and verified I could still read files from the drive. Note I was on Mac OSX Mojave and made sure to eject the drive.

Both Windows and Linux may cache FS changes that may lead to file system corruption if CircuitPython is disconnected or reset before everything has been flushed by the OS. There is nothing we can do about this since it is done host-side. We encourage folks to "safely eject" before removal and [use an editor that flushes immediately](https://learn.adafruit.com/welcome-to-circuitpython/creating-and-editing-code#editing-code).

Please comment if you can reproduce this still with a finite set of steps.

code.py
```python
import digitalio
import board
import time

led = digitalio.DigitalInOut(board.LED3)
led.direction = digitalio.Direction.OUTPUT
while True:
    led.value = not led.value
    time.sleep(0.1)
```

closed 2019-03-27 by tannewt