adafruit/circuitpython #1628: [nrf52840] Debug console no longer works

rdagger, 2019-03-10:
I'm using a Feather nRF52840 Express.  I have used it extensively for 2 weeks.  I coded a small game and everything was working great.  I upgraded to beta 4.3 and now the board is consistently crashing without giving any kind of debugging messages in the serial console.  This never happened prior to 4.3.  The crashing started instantly after the upgrade without any mods to my previously working code.  Prior to the upgrade I never experienced any crashing with the Feather nRF52840.

After crashing the drive and serial disappear.  I can some times get the serial back by launching safe mode.  I've tried storage.erase_filesystem() but the problem persists.  Unfortunately, the problem persists even after reverting back to beta 4.2.   I'm still not getting any helpful feedback from the serial console.  When the board crashes the NeoPixel gives a long green then a quick red and quick yellow (the pattern is repeating).

Occasionally before crashing I get "OSError: [Errno 5] Input/output error" and some times in safe mode I get "Crash into the HardFault_Handler."

Is there a way to reset the board back to factory?

dhalbert, 2019-03-10:
If you were running 4.0.0-beta.2 before you upgraded to beta.3, then reinstalling beta.2 would revert things to as they were: there's nothing "hidden" going on. It's possible a new bug was introduced in beta.3, or perhaps something about your operating conditions have changed. Please give us more information:

1. Source code of your program (attach as .zip or .txt)
2. More complete description of when the error happens. Can you reproduce it consistently, and how
3. Complete text of error when you get the `OSError`
4. What is your host computer operating system, what kind of port (USB, with or without hub, etc.)?

Thanks.

rdagger, 2019-03-10:
It turns out I did not revert far enough.  I loaded Adafruit CircuitPython 0dc2600 on 2019-02-20 and everything is now working!  Unfortunately, Adafruit CircuitPython f3e50b9 on 2019-02-21 does NOT work.
1.  Attached is the source code and a schematic.
[Mario Working 2.zip](https://github.com/adafruit/circuitpython/files/2949871/Mario.Working.2.zip)
2.  By “lock up”, I mean the program halts, the serial stops functioning and the feather drive disappears.  It happens consistently with f3e50b9 and never happens with 0dc2600
3.  Since the serial halts I’m not getting any error messages.  Occasionally, I've received the message "OSError: [Errno 5] Input/output error", but there was no other text.
4.  I'm running a Raspberry Pi 3.  The Feather board is connected with a 1ft USB cable.  I replaced the USB cable, all jumpers and tried a bench power supply.  I also confirmed 5V and 3.3V at the Feather using a meter.   Again everything works correctly with 0dc2600.

I am confident that the problem is due to a change that occurred between 0dc2600 and f3e50b9 .  It's likely a BLE or SPI issue.

rdagger, 2019-03-11:
I did further testing and I think the crash occurs on the first _spi.write()_ call.  This is line number 895 in method _write_cmd_cpy()_ of my file ssd1351.py.  Here's the line:
`self.spi.write(bytearray([command]))`

dhalbert, 2019-03-11:
Thanks - I am seeing the same place. Am debugging with gdb via J-Link.

dhalbert, 2019-03-12:
@rdagger could you test the attached .uf2? It works for me with your program (your program is very cute!). Thanks.

Problem was that nRF52840 SPIM3 peripheral got re-enabled in beta.3, but it has a hw bug, there's a workaround in the nrfx library, but the workaround breaks BLE. So disabled SPIM3. There was also another problem involving include files, but fixing that didn't fix the underlying SPIM3 problem.

[nrf52840-fix-spi-ble-2019-03-11.uf2.zip](https://github.com/adafruit/circuitpython/files/2954783/nrf52840-fix-spi-ble-2019-03-11.uf2.zip)

rdagger, 2019-03-12:
Thanks, great job!  The program works again using your .uf2.
Will your changes be incorporated into the next build?

dhalbert, 2019-03-12:
Thanks for testing. Yes, this fixes a significant bug, and will go into the next beta (and will be available in intermediate builds [here](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin) after the PR is merged)

PR #1638 (2019-03-12): nrf: Disable SPIM3; add SPIM1; remove TWIM1
SPIM3 has a hardware bug in rev1 nRF52840 (see errata). There's a workaround in nrfx, but it causes BLE to fail.

Reconfigured busio choices to use SPIM1 and SPIM2, disabling SPIM3. TWIM0 and TWIM1 are shared peripherals with SPIM0 and SPIM1, and it's not possible to shared dynamically right now due to static nrfx interrupt handler allocation. So choose to have one I2C bus available, and two SPI buses.

Later we may allow dynamic reconfig, try to fix the SPIM3 anomaly-fix bug (or at least report it with a small test case), and/or have some kind of co [... cut]
closed 2019-03-12 by tannewt