adafruit/circuitpython #1629: Is it Circuitpython support ESP32 now ?

water5, 2019-03-10:


dhalbert, 2019-03-10:
There is no native build for ESP32. However, we have a good library that supports using the ESP32 as a co-processor: https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI. This library is used for the PyPortal.

Using the ESP32 as a co-processor is advantageous in several ways: we don't need to port CircuitPython itself to the ESP32, which is a major task; we can substitute another Wifi-capable co-processor if another good one comes along, and having another core running the network code improves performance and removes potential concurrency issues.

microDev1, 2021-09-18:
CircuitPython just got a `BLE` workflow, so `esp32` support is now possible.

Neradoc, 2022-11-05:
The ESP32 is supported now via the Web Workflow.
I think we can close this issue.

closed 2022-11-05 by MicroDev1