adafruit/circuitpython #1630: Possible USB Serial performance prevent uPyCraft to upload files to CPX (CP4.0.0.beta2 & beta3)

barbudor, 2019-03-10:
Hi

I have recently upgraded my CPX to 4.0.0beta3.
I am working on a project where I need to log data to files on the file system (requires storage to be mounted as writable from boot.py).
In order to optimize the debugging I am using uPyCraft which is an IDE which allows to upload files to the file system through the REPL.
The code send by uPyCraft to the REPL is:
`myfile=open(filename,'w')
myfile.write( ...data... ) # data is a 128 byte string where end of line are all '\r'
.. repeat until whole file is sent ...
myfile.close()`

I noticed that file upload doesn't work anymore with either CP 4.0.0beta2 or beta3 while it is working fine with CP 3.1.2.

As it seems that the "protocol" breaks in the middle, my 1st assumptions is that a performance issue is blocking it.
Could you suggest some tests I could do to better analyse and provide a more useful issue description ?

Thanks

tannewt, 2019-03-11:
One helpful piece of information would be how big the file you are sending is. My guess is that the IDE is overflowing the CDC incoming buffer. I don't think we correctly tell USB to slow down.

barbudor, 2019-03-11:
Hi tannewt, 
It does not depends on the file size. Even a small file such as boot.py failed.
When I look at the uPyCraft code (https://github.com/DFRobot/uPyCraft_src/blob/master/ctrl.py line 1044), I understand that:
- File to write is cut in chunks of 128 bytes
- Each chunk is written with a `myfile.write(...data..)` REPL command
- After each write, they wait for REPL response. (They do not explicitly check for `">>>"` but they test for `"Traceback"` or `"..."` to decide for a failed transfer )
So I feel that file size would have no impact.

You are refering to the CDC incoming buffer. May I ask what is its length and if it changed between v3.1.2 and v4.0.0 ?

It is a bit difficult for me to qualify more precisely the issue. May be I should try to run uPyCraft from the source code instead of from the executable and try to debug it more precisely.

Thanks

tannewt, 2019-03-12:
The incoming buffer has changed with 4.0.0 because we moved from the Atmel USB stack to TinyUSB. 

Looks like the incoming buffer is 128 bytes on the SAMD21: https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/Makefile#L91

One easy fix would be to increase that value and see if the problem goes away. It will reduce the available heap memory so it's a balancing act.

barbudor, 2019-03-12:
Thanks tannewt
This clarify why "myfile( ...128 bytes...)" (obviously slightly larger than 128 bytes) fails.

In the meantime I found a menu in uPyCraft to force the board to be considered as a microbit which enforce a smaller transfer chunk (64 bit). Forcing this mode seems to make the file transfer reliable.
I was about to report this tonight.

May be we could add a short comment in https://circuitpython.readthedocs.io/en/latest/ports/atmel-samd/README.html (Connecting ... serial...) to state the buffer length ?

We can close the issue.
Many thanks

tannewt, 2019-03-13:
We should be able to change the USB stack so that host-side software doesn't need to know the buffer size. I'm going to leave this open and edit the title. We need to NAK when the CDC buffer is full.

tannewt, 2019-03-21:
I can easily reproduce this by pasting a long string into the REPL via screen. Will try and fix shortly.

tannewt, 2019-03-21:
TinyUSB PR here: https://github.com/hathach/tinyusb/pull/46

PR #1710 (2019-03-27): Update TinyUSB to fix CDC OUT buffering
Fixes #1630
commit 1b973254 (2019-03-27): Update TinyUSB to fix CDC OUT buffering

Fixes #1630
closed 2019-03-27 by dhalbert