adafruit/circuitpython #1636: Idea: hide "private" names from tab completion

deshipu, 2019-03-11:
This is just an idea for discussion: what if we filtered the names considered by auto-completion to filter out all names that start with the underscore, *unless* the user has already typed an underscore? The rationale behind this is that those usually are implementation details of the module/class in question, and the chance the users actually want to call them from outside are small.

tannewt, 2019-03-12:
Makes sense to me. I thought we tab completed with the entered prefix but I haven't looked into it.

deshipu, 2019-03-12:
We do. This is only for the case when the prefix is empty (as right after a period, for example).

katlings, 2019-05-06:
This sounds fun; I'm going to tackle it for PyCon 2019 sprints!

tannewt, 2019-05-09:
Done by @katlings with #1850!

closed 2019-05-09 by tannewt