adafruit/circuitpython #1642: Linux Ubuntu 18.04 USB Connection Fail

tboie, 2019-03-13:
3.x - OK
4.x - Not working

UF2 Bootloader v2.0.0-adafruit.9 SFHWRO
Model: Feather M0 Express
Board-ID: SAMD21G18A-Feather-M0-Express-v0

User: root

Initial USB connection dmesg:
[  212.866387] usb 6-1: new full-speed USB device number 2 using ohci-platform
[  213.058283] usb 6-1: New USB device found, idVendor=239a, idProduct=8023
[  213.058317] usb 6-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[  213.058338] usb 6-1: Product: Feather M0 Express
[  213.058358] usb 6-1: Manufacturer: Adafruit Industries LLC
[  213.058378] usb 6-1: SerialNumber: 80C840FA0535E405A413E213D01150FF
[  213.101387] usb 6-1: ctrl urb status -62 received
[  213.106331] input: Adafruit Industries LLC Feather M0 Express as /devices/platform/fe3a0000.usb/usb6/6-1/6-1:1.3/0003:239A:8023.0006/input/input6
[  213.164897] hid-generic 0003:239A:8023.0006: input,hidraw5: USB HID v1.11 Keyboard [Adafruit Industries LLC Feather M0 Express] on usb-fe3a0000.usb-1/input3
[  213.168839] usb 6-1: USB disconnect, device number 2
[  213.230462] usbcore: registered new interface driver cdc_acm
[  213.230477] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  213.249278] usbcore: registered new interface driver usb-storage
[  213.257665] usbcore: registered new interface driver uas
[  214.503344] usb 6-1: new full-speed USB device number 3 using ohci-platform
[  214.694671] usb 6-1: New USB device found, idVendor=239a, idProduct=8023
[  214.694706] usb 6-1: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[  214.694727] usb 6-1: Product: Feather M0 Express
[  214.694748] usb 6-1: Manufacturer: Adafruit Industries LLC
[  214.694768] usb 6-1: SerialNumber: 80C840FA0535E405A413E213D01150FF
[  214.700275] cdc_acm 6-1:1.0: ttyACM0: USB ACM device
[  214.706065] usb-storage 6-1:1.2: USB Mass Storage device detected
[  214.708020] scsi host0: usb-storage 6-1:1.2
[  224.729735] hid-generic 0003:239A:8023.0007: timeout initializing reports
[  224.732838] input: Adafruit Industries LLC Feather M0 Express as /devices/platform/fe3a0000.usb/usb6/6-1/6-1:1.3/0003:239A:8023.0007/input/input7
[  224.791960] hid-generic 0003:239A:8023.0007: input,hidraw5: USB HID v1.11 Keyboard [Adafruit Industries LLC Feather M0 Express] on usb-fe3a0000.usb-1/input3
[  247.282391] usb 6-1: reset full-speed USB device number 3 using ohci-platform
[  252.440373] usb 6-1: device descriptor read/64, error -110
[  267.697900] usb 6-1: device descriptor read/64, error -110
[  267.956923] usb 6-1: reset full-speed USB device number 3 using ohci-platform
[  273.115081] usb 6-1: device descriptor read/64, error -110
[  288.374190] usb 6-1: device descriptor read/64, error -110
[  288.632093] usb 6-1: reset full-speed USB device number 3 using ohci-platform
[  299.035480] usb 6-1: device not accepting address 3, error -110
[  299.193315] usb 6-1: reset full-speed USB device number 3 using ohci-platform

Bootloader dmesg:
[ 1614.625615] usb 6-1: new full-speed USB device number 8 using ohci-platform
[ 1614.820757] usb 6-1: New USB device found, idVendor=239a, idProduct=001b
[ 1614.820791] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1614.820813] usb 6-1: Product: Feather M0 Express
[ 1614.820833] usb 6-1: Manufacturer: Adafruit Industries
[ 1614.820853] usb 6-1: SerialNumber: AF048C08504E5350312E314AFF05110D
[ 1614.823339] cdc_acm 6-1:1.0: ttyACM0: USB ACM device
[ 1614.826938] usb-storage 6-1:1.2: USB Mass Storage device detected
[ 1614.828425] scsi host1: usb-storage 6-1:1.2
[ 1614.838833] hid-generic 0003:239A:001B.0008: hiddev0,hidraw5: USB HID v1.00 Device [Adafruit Industries Feather M0 Express] on usb-fe3a0000.usb-1/input3
[ 1615.836969] scsi 1:0:0:0: Direct-Access     Feather  M0 Express       1.00 PQ: 0 ANSI: 2
[ 1615.848840] sd 1:0:0:0: [sda] 16000 512-byte logical blocks: (8.19 MB/7.81 MiB)
[ 1615.854808] sd 1:0:0:0: [sda] Write Protect is off
[ 1615.854848] sd 1:0:0:0: [sda] Mode Sense: 0f 00 00 00
[ 1615.860803] sd 1:0:0:0: [sda] No Caching mode page found
[ 1615.861378] sd 1:0:0:0: [sda] Assuming drive cache: write through
[ 1615.902830]  sda:
[ 1615.943968] sd 1:0:0:0: [sda] Attached SCSI removable disk
[ 1684.757806] FAT-fs (sda): utf8 is not a recommended IO charset for FAT filesystems, filesystem will be case sensitive!

cpforbes, 2019-03-13:
This looks similar to #1617

caternuson, 2019-03-15:
Just adding info. I'm not able to recreate this. Ubuntu 18.04.2 LTS. CircuitPython 4.0.0 beta 3:
`adafruit-circuitpython-feather_m0_express-en_US-4.0.0-beta.3.uf2`
on a Feather M0 Express.

```console
[Fri Mar 15 13:42:29 2019] usb 2-1.7: new full-speed USB device number 89 using ehci-pci
[Fri Mar 15 13:42:29 2019] usb 2-1.7: New USB device found, idVendor=239a, idProduct=8023
[Fri Mar 15 13:42:29 2019] usb 2-1.7: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[Fri Mar 15 13:42:29 2019] usb 2-1.7: Product: Feather M0 Express
[Fri Mar 15 13:42:29 2019] usb 2-1.7: Manufacturer: Adafruit Industries LLC
[Fri Mar 15 13:42:29 2019] usb 2-1.7: SerialNumber: E879706F1475D4050213E2739271F0FF
[Fri Mar 15 13:42:29 2019] cdc_acm 2-1.7:1.0: ttyACM0: USB ACM device
[Fri Mar 15 13:42:29 2019] usb-storage 2-1.7:1.2: USB Mass Storage device detected
[Fri Mar 15 13:42:29 2019] scsi host6: usb-storage 2-1.7:1.2
[Fri Mar 15 13:42:29 2019] input: Adafruit Industries LLC Feather M0 Express as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.7/2-1.7:1.3/0003:239A:8023.0051/input/input62
[Fri Mar 15 13:42:29 2019] hid-generic 0003:239A:8023.0051: input,hidraw4: USB HID v1.11 Keyboard [Adafruit Industries LLC Feather M0 Express] on usb-0000:00:1d.0-1.7/input3
[Fri Mar 15 13:42:30 2019] scsi host6: scsi scan: INQUIRY result too short (5), using 36
[Fri Mar 15 13:42:30 2019] scsi 6:0:0:0: Direct-Access     Adafruit Feather M0 Expre 1.0  PQ: 0 ANSI: 2
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: Attached scsi generic sg3 type 0
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: [sdb] 4089 512-byte logical blocks: (2.09 MB/2.00 MiB)
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: [sdb] Write Protect is off
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: [sdb] Mode Sense: 03 00 00 00
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: [sdb] No Caching mode page found
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: [sdb] Assuming drive cache: write through
[Fri Mar 15 13:42:30 2019]  sdb: sdb1
[Fri Mar 15 13:42:30 2019] sd 6:0:0:0: [sdb] Attached SCSI removable disk
[Fri Mar 15 13:42:31 2019] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
```

@tboie Do you have udev rules installed?

tboie, 2019-03-15:
Yes, using these:
https://learn.adafruit.com/adafruit-arduino-ide-setup/linux-setup

dhalbert, 2019-03-15:
Just checking: have you rebooted this machine and it still doesn't work? We saw some issues with Arch Linux in which an update was done and there were problems before the machine was rebooted.

tboie, 2019-03-16:
Yes, no luck

dhalbert, 2019-03-16:
What is the brand and model of the PC that you're using here, and is your USB port on the motherboard, or is it a separate plugin card? I notice that it's using the "ohci-platform" driver, rather than echi or xhci. I believe this is older.

If you have a USB hub you could interpose between the host's USB port and the board, try that.

tboie, 2019-03-17:
Pine 64 - RockPro64:
Rockchip RK3399 Hexa-Core (dual ARM Cortex A72 and quad ARM Cortex A53) 64-Bit Processor with MALI T-860 Quad-Core GPU

https://www.pine64.org/?product=rockpro64-4gb-single-board-computer

Ubuntu 18.04.2 Bionic

On-board USB 2.0 connection being used.

CIRCUITPY 3.x is working

dhalbert, 2019-03-17:
I did a little research on this board. There have been USB problems in the past, though they seem to be related to USB 3.0. Have you tried the USB 3.0 ports, or the USB-C port with an adapter?

I'm suspecting there's something marginal about the USB driver or ports. If you could get some USB traces with wireshark (if that works) or a Beagle USB tracer, then we'd have a starting point for what's going on. Otherwise, I'm not sure we have anything to offer - sorry about that.

tboie, 2019-03-21:
All USB ports have been tested with no success.  Tested dozens of other USB items with no problems.

The new USB midi is just too cool for me to give up just yet.

I have wireshark running.  Is there any particular method to get the needed log information?

dhalbert, 2019-03-21:
To set up wireshark to trace USB traffic, follow the directions here: https://wiki.wireshark.org/CaptureSetup/USB. The critical thing is doing `sudo modprobe usbmon`. You could then run wireshark as root and not do the other stuff, but it's easier if you're not root.

When you start wireshark with usbmon loaded, you should see a list of usb controllers and see usb activity (there's a line graph showing activity), like this:

[image]

You'l see activity on one line from mouse movement (try moving your mouse). Plug in the board, and see which line shows activity.  If you can use a separate controller, great, but there may be only one, so your keyboard and mouse activity will be mixed in. Double-click the appropriate "usbmon", then start a capture (it may start automatically when you double-click), plug in the board, and wait say 30 seconds or so, and then stop the capture. Then save the data with "Save As...", as a .pcapng file. You can then zip it up and post it here.

This kind of data is invaluable for debugging. Thanks!

tboie, 2019-03-22:
[dump1.pcapng.zip](https://github.com/adafruit/circuitpython/files/2998717/dump1.pcapng.zip)

Hope that helps.  Can run more captures if needed

dhalbert, 2019-03-23:
Thanks!

@hathach ^^ here's another USB trace, from a different host computer with issues (it's a Rockchip running Ubuntu).

dhalbert, 2019-03-28:
@tboie Could you try this build on your Feather M0 Express?
https://adafruit-circuit-python.s3.amazonaws.com/bin/feather_m0_express/en_US/adafruit-circuitpython-feather_m0_express-en_US-20190328-c6e3b17.uf2.

It fixes a USB HID problem that may also affect connectivity. PR for fix is #1721.

tboie, 2019-03-28:
Works after a quick test!

dhalbert, 2019-03-28:
OK, we'll declare this fixed by #1721!

closed 2019-03-28 by dhalbert